RetroPie/RetroPie-Setup #1202: Registering Bluetooth controller with RetroPie 3.4

biscuits99, 2016-01-22:
- Loaded 3.4 image to microSD card
- Booted into ES, configured keyboard
- Expanded disk space via raspi-config in ES, rebooted
- In ES chose retropie setup and updated script (seemed very picky about connecting to github, but worked first time via remote PuTTY session, anyway... it updated)
- Went into setup menu and bluetooth menu
- Saw my 8bitdo FC30 Pro fine, I select option 1 DisplayYesNo but at this step pictured just hangs

[image]

I'll try to do it manually on a new clean 3.4

joolswills, 2016-01-22:
Some incompatibility perhaps - After the scan is done, I would have expected it to be able to connect.

You can try using "bluetoothctl" to manually pair

`
agent DisplayYesNo
scan on
pair MACADDRESS
connect MACADDRESS
`

etc

biscuits99, 2016-01-22:
Yep - manually pairing works a treat. Very close to getting this working smoothly using the retropie bluetooth module. Few more tests and I'll update ticket.

biscuits99, 2016-01-22:
Ok, I have it working now so that my 8bitdo controllers are connected automatically at startup. I couldn't get the process to run reliably with the inbuilt bluetooth menu item though - which is more about me not understanding how it works than there being an issue with it.

Essentially I did the following

- bluetoothctl
- power on
- scan on
- agent on
- pair [MAC Address]
- trust [MAC Address]
- connect [MAC Address]
- exit

Then created a script to run in /etc/init.d/

```
#!/bin/bash
sudo bluetoothctl << EOF
power on
connect 01:06:A1:62:61:07
exit
EOF
```

and created a udev rule as per:
https://github.com/RetroPie/RetroPie-Setup/issues/1185#issuecomment-172127933

I could write up a guide for this, or is it worth waiting to see the relevant logic can be incorporated into the retropie bluetooth menu option?

Side note: The restart from ES seems to take about 4 minutes.

joolswills, 2016-01-23:
The poster on the issue you linked to was able to register / pair their controller with our script fine - so maybe it's related to bluetooth dongle etc. Works fine for my keyboard also. Their controller also pairs automatically on start I think, so don't know why you need the extra code - different firmwares etc ?

Improvements will be made to the bluetooth pairing for jessie, and I could movemore of it over to bluetoothctl, but it isn't designed for automation much like the bluez test scripts and it's more involved than just feeding it some data as you have done - eg if it replies to enter a code into a keyboard, we would need to capture that.

The ES restart is not a bug - it was broken before : https://github.com/RetroPie/RetroPie-Setup/issues/1200

biscuits99, 2016-01-23:
When I was trying above, on occasion the bluetooth menu script did work without issue, and connect the FC30 Pro, but often also produce the error in the first post (and never worked first time). I couldnt work out what caused it to work sometimes but not others.
Although I was never able to get it to automatically re-connect on reboot.

Yes, good point about the bluetooth dongle, I'll try a couple of others I have, and I'll also try the exact controller he was using. I'll double check all firmware is current first.

Sorry, didnt notice the report about the ES restart issue, if relevant I hadnt loaded any extra roms to the system.

biscuits99, 2016-01-23:
Ok, tested SFC30 on firmware 2.65 (Start+R) - works pretty flawlessly. Although controller codes generated dont match correctly.

Attached the output files (driver is udev as per default set).
[8BitdoSFC30GamePadJoystick.txt](https://github.com/RetroPie/RetroPie-Setup/files/101851/8BitdoSFC30GamePadJoystick.txt)
[es_input.txt](https://github.com/RetroPie/RetroPie-Setup/files/101850/es_input.txt)
[es_temporaryinput.txt](https://github.com/RetroPie/RetroPie-Setup/files/101852/es_temporaryinput.txt)

I manually edited the controller file with these details and it works fine:
https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/8Bitdo-SFC30-GamePad-Bluetooth-Mode(START%2BR).cfg

Will now try same with FC30 Pro

biscuits99, 2016-01-23:
It works! Including auto reconnecting.
It seems the FC30 Pro is very picky about the point it attempts to connect, but turning it on just prior to the retropie splashscreen works well. It doesnt seem to consistently appear in the detected devices list (as well as throw that dbus error from time to time) when registering in the bluetooth menu, but a few retries seems to get it to appear if it doesnt at first.

Also, its controller file is generated correctly without issue.
Thanks for your help @joolswills

Lokagan, 2016-01-23:
For myself, i setup raspbian and use the retropie setup script (yes i want desktop mode). My FC30pro is successfully associating (and autoreconnect) with jessie. Emulation Station & Retroarch didn'see it even if jstest return perfectly. Any idea ? SDL2 issue ?

biscuits99, 2016-01-23:
This may help you:
https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller#step-2---manual-file-edit-for-8bitdo-controllers

Gaander, 2016-01-24:
Nice! my NES30 Pro finaly works now!
But I still get a lot of errors/Infos in the background when ES or a Game is running, like:
`INFO: The key you just pressed is not recognized by SDL. To help get this fixed, please report this to the SDL mailing list <mailaddress_of_the_mailing_list> EVDEV KeyCode 304`
The KeyCode varies.

Lokagan, 2016-01-25:
Adding the rule ENV{ID_INPUT_JOYSTICK}="1" did the trick, thanks biskcuits99.
The SDL infos have always been present, don't care about it... it works :)
For those who have random connection issue, putting your bluetooth adapter in discoverable mode make it works greater

closed 2016-03-21 by joolswills