RetroPie/RetroPie-Setup #1204: Copyright of Screenshots in the Wiki?

edent, 2016-01-24:
I'm writing an article about RetroPie for a magazine and I'd like to include some of the set-up screenshots from your the [Wiki](https://github.com/RetroPie/RetroPie-Setup/wiki/First-Installation)

For example https://cloud.githubusercontent.com/assets/10035308/9140482/cf42f25c-3cee-11e5-8f91-c1fc1c57175c.png

I appreciate that the source code is covered by the GPL, but I wondered if the documentation was covered by something else.

Thanks :-)

InsecureSpike, 2016-01-24:
hi, if you need any of my theme screen shots they can be found

https://insecurespike.wordpress.com/clean_canela-screenshots/

use as you like

joolswills, 2016-01-24:
Please go ahead and use the images. Thanks.

edent, 2016-01-24:
Thanks - much appreciated.

closed 2016-01-24 by edent