RetroPie/RetroPie-Setup #1211: USB ROM Copy on PI 2 Running Version 3.4

taylorbat, 2016-01-28:
In order to automate the ROM transfer process I have created a USB thumb drive with all of my ROMs in the correct folder structure. In previous versions after expanding the file system, restarting, and then inserting the USB drive the system would begin copying the ROMs within about 15 or 20 min. In this current version it appears that the ROMs will copy in the beginning of the reboot process for a few minutes and then stop. Currently the only way to transfer the entire USB drive of games is to restart the entire system over and over until all games have been transferred. Is anyone else having this issue? Has the USB ROM file transfer process changed in this version?

joolswills, 2016-01-28:
I have heard others mention it on the forum. Nothing has changed in our scripts, so it must be related to the underlying os. Possibly something has changed in usbmount.

taylorbat, 2016-01-28:
The last time this worked for me it worked on 3.3.1 wheezy, I dont have a copy of 3.3.1 jessie to test it.

joolswills, 2016-02-05:
I have replicated the problem. The problem is caused by systemd on Jessie killing the script - need to look into disabling the timeout

```
Feb  5 07:47:51 retropie usbmount-01_retropie_copyroms-[935]: Attempting to create directory structure for ROMS in '/media/usb0/retropie/roms' ...
Feb  5 07:47:54 retropie usbmount-01_retropie_copyroms-[935]: Syncing roms ...
Feb  5 07:47:54 retropie usbmount-01_retropie_copyroms-[935]: Syncing configs ...
Feb  5 07:48:20 retropie systemd-udevd[848]: timeout: killing '/usr/share/usbmount/usbmount add' [878]
Feb  5 07:48:20 retropie systemd-udevd[848]: '/usr/share/usbmount/usbmount add' [878] terminated by signal 9 (Killed)
Feb  5 07:48:39 retropie kernel: [   83.214594] usb 1-1.4: USB disconnect, device number 6
Feb  5 07:48:39 retropie systemd-udevd[1040]: inotify_add_watch(6, /dev/sda, 10) failed: No such file or directory
Feb  5 07:48:39 retropie usbmount[1044]: executing command: umount -l /media/usb0
Feb  5 07:48:39 retropie kernel: [   83.493965] FAT-fs (sda1): unable to read boot sector to mark fs as dirty
Feb  5 07:48:39 retropie usbmount[1044]: executing command: run-parts /etc/usbmount/umount.d
```

joolswills, 2016-02-05:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=740142

joolswills, 2016-02-05:
Looks like a problem in the way usbmount is implemented and we may need to use another mechanism for this - which is rather annoying as it all worked before systemd.

joolswills, 2016-02-05:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=658028

joolswills, 2016-02-05:
https://github.com/hfuchs/usbmount/tree/hfuchs.658028

joolswills, 2016-02-05:
I will include the fixed above usbmount package with the image for now to hopefully sort things.

joolswills, 2016-02-05:
the correct way is to launch a systemd service - the "fixed" usbmount doesn't work either as it get killed by udev still.

joolswills, 2016-02-05:
I am going to have to disable the non working usbmount for now until we can roll our own working solution that uses udev + systemd service. usbmount is just broken by design unfortunately

joolswills, 2016-02-05:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=774149 - contains working systemd service we can use

joolswills, 2016-02-05:
I have implemented a hopefully working usbmount for jessie now

please update retropie-setup and then go to the setup menu and choose to enable the usbromservice again. it should go and grab/build a package, and hopefully then it will work.

bernd69, 2016-02-05:
Had the same problem. Now I´ve tried your fix, but it doesn´t work for me.
- disabled usbromservice
- enabled usbromservice
- put in again an empty FAT formatted stick (which worked before), waited several time, but there is no folder "retropie/roms" folder setup.

Actually i only have the 3.4 image. Where to donload to older ones like 3.3.1 ? Cause actually I´m setting up a new system with a PI2.

Thanks and best regards
Boern

joolswills, 2016-02-05:
did you update retropie-setup first?

The changes have been tested by two people so I'm pretty sure they work, so let's debug it before you downgrade :)

joolswills, 2016-02-05:
it doesn't use an empty stick btw,. you need to make a retropie folder first in the root.

bernd69, 2016-02-05:
Sorry. Forget it. I have two Sandisk stick which are the same. One is formatted with Mac OSX. I´ve scrambled them ...

Thanks, Bernd

HerbFargus, 2016-02-07:
Issue should be resolved now.

closed 2016-02-07 by HerbFargus