RetroPie/RetroPie-Setup #1212: Replacement version of Linapple

dabonetn, 2016-01-28:
I've made a lot of changes to linapple to better support frontends.

I'd like to get it included in retropie.

It now supports direct launching of floppy images, and joystick remapping and quitting from joystick.
It also looks in the users home directory .linapple directory for the linapple.conf and Master.dsk.

Project is located here.

https://github.com/dabonetn/linapple-pie

For integration into retropie that I see the following changes needed.

Change to needed files to the following

        'CHANGELOG'
        'INSTALL'
        'LICENSE'
        'linapple'
        'Master.dsk'
        'README'
	'README-linapple-pie'
	'linapple.conf'

Make a symbolic link from /home/pi/.linapple to /opt/retropie/configs/apple2

Place the linapple.conf and Master.dsk into /opt/retropie/configs/apple2

Change the startup scripts to reflect the command line changes.

(linapple -1 %ROM%)

Change the media types to refect apple disk images (.po .PO .dsk .DSK .nib .NIB)

Get controller mappings from the main config into the linapple.conf file.

Apple joysticks only have max 2 buttons on joystick one, and one button on joystick two.

The following is from the linapple.conf file, I tried to make it as simple as possible to reconfigure the joysticks.

#######################################################################
#
#	Joysticks, those sticks of joy! There may be 2 joysticks at the same time
# Possible values are:
#	0	- joystick disabled
#	1	- use PC joystick #1 or #2, for corresponding joystick

#	2	- Keyboard standard
#	3	- Keyboard centered
# When joysticks used as a keyboard, they are stuck to Numpad keys (1-9 - axis movement, 0 - button1, . - button2)
#  when centered used, axis of joystick will be centered after releasing any cursor (Numpad 1..9) key.
# otherwise it assumed to be pressed always

#	4	- Use mouse as a joystick. Rather interesting thing, try it. Useful in Fantavision(tm)by Broderbund Software

# Default values are 2 for Joystick 0, and 0 for Joystick 1

	Joystick 0	=	1
	Joystick 1	=	0

# For Joysticks you can define which Joystick index number, axis number, and buttons.
# 
# Default for Joystick 1 is index 0, axis 0 and 1, buttons 0 and 1.
# Default for Joystick 2 is index 1, axis 0 and 1, button 0.

	Joy0Index	=	0
	Joy1Index	=	1
	Joy0Button1	=	0
	Joy0Button2	=	1
	Joy1Button1	=	0
	Joy0Axis0	=	0
	Joy0Axis1	=	1
	Joy1Axis0	=	0
	Joy1Axis1	=	1
	
# Enable Quitting the program with by pressing two joystick buttons at the same time
# Default is disabled, default buttons are 8 and 9.

	JoyExitEnable	=	0
	JoyExitButton0	=	8
	JoyExitButton1	=	9

joolswills, 2016-01-29:
Thanks for that. Looks good - will switch adjust the script when I next do some dev. Cheers.

HerbFargus, 2016-05-02:
added as of https://github.com/RetroPie/RetroPie-Setup/pull/1437

closed 2016-05-02 by HerbFargus