RetroPie/RetroPie-Setup #1213: 8Bitdo NES30 Controller - Configuring not working

henn1nk, 2016-01-28:
Hello guys, i am running the controller in Mode1 http://www.nes30.com/manual.html and was able to pair it via retropie_setup.sh.. Starting retropie the gamepad is detected, but i am not able to setup the D-PAD keys.. Any idea why?

Need i to choose another mode? Which mode should i use for two nes30 controllers?

henn1nk, 2016-01-29:
With mode2 i have problems to pair:

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/home/osmc/RetroPie-Setup/scriptmodules/supplementary/bluetooth/bluez-simple-agent", line 118, in pair_reply
    dev_connect(dev_path)
  File "/home/osmc/RetroPie-Setup/scriptmodules/supplementary/bluetooth/bluez-simple-agent", line 38, in dev_connect
    dev.Connect()
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.NotAvailable: Operation currently not available

henn1nk, 2016-01-29:
using sudo bluetoothctl:

```
[NEW] Controller 00:1A:7D:DA:71:05 Wohnzimmer [default]
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:1A:7D:DA:71:05 Discovering: yes
[NEW] Device 01:01:4F:67:79:0A 8Bitdo NES30 GamePad
[bluetooth]# pair 01:01:4F:67:79:0A
Attempting to pair with 01:01:4F:67:79:0A
[CHG] Device 01:01:4F:67:79:0A Connected: yes
[CHG] Device 01:01:4F:67:79:0A Modalias: usb:v2820p0009d0100
[CHG] Device 01:01:4F:67:79:0A UUIDs:
	00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 01:01:4F:67:79:0A Paired: yes
Pairing successful
[CHG] Device 01:01:4F:67:79:0A Connected: no
[bluetooth]# trust 01:01:4F:67:79:0A
[CHG] Device 01:01:4F:67:79:0A Trusted: yes
Changing 01:01:4F:67:79:0A trust succeeded
[bluetooth]# connect 01:01:4F:67:79:0A
Attempting to connect to 01:01:4F:67:79:0A
[CHG] Device 01:01:4F:67:79:0A Connected: yes
Failed to connect: org.bluez.Error.NotAvailable
[CHG] Device 01:01:4F:67:79:0A Connected: no
[CHG] Device 01:01:4F:67:79:0A RSSI: -69
[CHG] Device 01:01:4F:67:79:0A RSSI: -61
```

joolswills, 2016-01-30:
Please try the forum first as other users may be able to help https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

wraybowling, 2016-02-13:
I have been getting the same sort of errors from all the bluez tools. There are a few things I can think of that I did.. #1 was installing a PS3 controller before the 8bitdo controller on 3.2. #2 was going through the bluetooth connection menu while the 8bitdo controller was still plugged in. Neither of these methods I can confirm are a source of trouble.. We'd have to flash the latest versions of everything and try to make the bug happen again to be sure.

orangenpresse, 2016-02-19:
https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller

and use the
Pre-Generated retroarch controller files, saves a lot of time
https://github.com/libretro/retroarch-joypad-autoconfig/tree/master/udev

closed 2016-03-26 by joolswills

RadBrad87, 2017-01-20:
I had the following error that occurs if you configure a PS3 controller before attempting to configure an 8bitdo gamepad:

`Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/home/osmc/RetroPie-Setup/scriptmodules/supplementary/bluetooth/bluez-simple-agent", line 118, in pair_reply
    dev_connect(dev_path)
  File "/home/osmc/RetroPie-Setup/scriptmodules/supplementary/bluetooth/bluez-simple-agent", line 38, in dev_connect
    dev.Connect()
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.NotAvailable: Operation currently not available`

The solution is to remove the PS3 controller driver via the retropie_setup.sh script ([Manage packages] -> [driver]->[ps3controller]->[uninstall/remove]).

Afterward, manually remove all packages/dependencies related to it. Basically any package mentioned in the PS3 controller setup guide:

https://github.com/retropie/retropie-setup/wiki/Setting-up-a-PS3-controller

`sudo apt-get --purge remove sixad`
`sudo apt-get --purge remove sixpair`
`sudo apt-get --purge remove bluetooth`
`sudo apt-get --purge remove blueman`
`sudo apt-get --purge remove bluez-hcidump`
`sudo apt-get --purge remove checkinstall`
`sudo apt-get --purge remove libusb-dev`
`sudo apt-get --purge remove libbluetooth-dev`
`sudo apt-get --purge remove joystick`
`sudo apt-get --purge remove pkg-config`
`sudo apt-get --purge remove python-gobject`
`sudo apt-get --purge remove bluez-utils`
`sudo apt-get --purge remove pi-bluetooth`

Then,

`sudo apt-get  install pi-bluetooth`

Then follow the normal 8bitdo setup guide:
https://github.com/retropie/retropie-setup/wiki/Setting-up-an-8bitdo-Bluetooth-controller

joolswills, 2017-01-20:
@RadBrad87 when installing the ps3 controller from retropie-setup it says

>NOTE: You cannot currently use PS3 controllers with other bluetooth devices. The PS3 controller driver disables the standard bluetooth stack. If you want to use a wireless keyboard along with your PS3 controllers you can use 2.4ghz wireless devices that come with their own dongle.

you don't need to purge so many packages though - removal of the driver from retropie-setup should be enough to restore bluetooth. The forum is a better place for discussion of this (this ticket is closed).