RetroPie/RetroPie-Setup #1215: Favorites List and/or Recently Played List

Axeavius, 2016-01-29:
I just had a great idea for a new feature in EmulationStation. So, I went over to emulationstation.org/forum, signed up, and found a thread for the same idea I had. But unfortunately, the entire forum is like a zombie spambot apocalypse!

I searched Facebook and found nothing. However, /r/emulationstation on Reddit is still active, so I'm going to make a post about it there as well.

Anyway, the feature request I (and others) had thought of was for a Favorites List and/or a Recently Played list. Someone on the other thread even brought up the idea of making the Favorites List coalesce the favorites for all emulators into one list.

Keep up the good work, you guys...rock!

joolswills, 2016-01-30:
https://github.com/RetroPie/EmulationStation/issues for ES related features - we may  add something like that in the future, but the more we develop ES (and I certainly dont really want to take on a project like that), the further it gets from upstream - we are hopeful that upstream development will continue at some point.

tynor88, 2016-01-31:
@joolswills The last ES github commit was 8th march 2015 - i doubt it will be picked up again. There are many great pull requests waiting to be merged in, but they are all stalled unfortunately. I think it's time for someone to start a new repo so we can have all the good features merged in. 

@Axeavius as for the feature to maybe a favourite list it's actually already added - but not merged in. [See here ](http://blog.petrockblock.com/forums/topic/kid-friendly-retropiees-ui-modes-favorites-hiding-items-s-testers-wanted/) - @gizmo98 made a pull requests to add that feature. I'm missing it too.

joolswills, 2016-01-31:
Still under a year - I've left projects for longer and come back to them.

We will likely merge in a few more improvements, but that's not the same as maintaining ES - plus our fork is specific to RetroPie with the input configuration. The Recalbox fork is also actively developed, but again with some features specific to their project. I still hope Aloshi will find time to continue at some point.

closed 2016-03-21 by joolswills

jerkball, 2017-02-28:
Not yet a year closing the thread, but already over a year since last post ;)

Common, browsing through a complete romset is pain in the ashes...

And creating different dirs for your favorite roms is so 90s... your kids ask you for fever...

deurk, 2017-03-16:
Recalbox implemented it, would be nice to see it in Retropie, even if it is an ES feature.