RetroPie/RetroPie-Setup #1217: /opt/ vs. /etc/

nschloe, 2016-01-30:
In some areas, RetroPie does things somewhat differently than what you would expect by a Linux software. For example, it installs configuration into `/opt/`. Are more common directory would probably be
```
/etc/retropie/
```

joolswills, 2016-01-30:
We prefer to keep everything together as much as we can- I think as something not shipping with the distribution using /opt is fine, http://www.tldp.org/LDP/Linux-Filesystem-Hierarchy/html/opt.html

It's not something we would change now anyway - as it would be a pain for previous installs etc.

joolswills, 2016-01-30:
I believe if following "standards" /etc/opt/retropie would be the proper place btw, but this won't change as mentioned. you can always symlink it though if you want to access it via that path.

nschloe, 2016-01-30:
I had been under the impression that RetroPie scatters its date throughout the filesystem anyways. There's  the stuff in `$HOME`, for example.

The question of keeping everything that's user-defined in one place came up to me just now when I decided to update RetroPie. I'll have to back-up everything that's mine, but to be honest, I can't quite recall. The controller configs in `/opt/something/something/something/dark_side`, the roms in `$HOME/something/snes/whatever`; perhaps something else?

Having those files in one place would actually allow for a smoother backup/rollback/upgrade for users.

joolswills, 2016-01-30:
some of the things that come with RetroPie do use $HOME, and it would require significant work to change the config locations of everything we ship with. we do though sy,link everything from $HOME to /opt/retropie/configs - there shouldn't be anything that actually is in $HOME apart from the syllinks. There are a few init.d things also, which we dont have a choice about where they live.

only the rom foles themselves live in $HOME/RetroPie - this has always been the case since I tarted on the project - moving it isn't worth the effort of the problems people will have, and the work involved.

joolswills, 2016-01-30:
fixed typos - should put my glasses on.

nschloe, 2016-01-30:
> `init.d`

Those aren't user files, I trust.

What are the config files I need to backup such that when I put them back into a new system, I get back everything I had configured?

joolswills, 2016-01-30:
they are not considered user editable no.

/opt/retropie/configs

nschloe, 2016-01-30:
+ all of `$HOME/RetroPie`?

joolswills, 2016-01-30:
yeh and your roms, sorry.

nschloe, 2016-01-30:
The roms and splash screens should all be in `$HOME/RetroPie` I guess. Thanks!

closed 2016-01-30 by nschloe

joolswills, 2016-01-30:
user splashscreens live in $HOME/RetroPie/splashscreens if you have put some there.