RetroPie/RetroPie-Setup #1218: painful upgrade path

nschloe, 2016-01-30:
From what I understand, the recommended update path for RetroPie is the following:

  * Download an 800+MB zipped image,
  * Back up whatever data you have on your SD-Card,
  * wipe the card and put the new image on it,
  * put all your backed up data back on it.

Steps 2 and 4 are particularly painful for me as I'm using the Raspberry Pi for things other than RetroPie, too. Having a couple of hundred (or thousand?) users download such a large file from your server must be somewhat painful for the maintainers.

Is there an upgrade path that swaps in-and-out only the parts that are RetroPie's?

(That's for everything except a base-system upgrade of course.)

joolswills, 2016-01-30:
The forum is considered the place for support discussion btw - Please see https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md (which is linked when opening a new issue).

You can update like this https://github.com/retropie/RetroPie-Setup/wiki/Updating-RetroPie

joolswills, 2016-01-30:
Moving from from Wheezy to Jessie a re-image is best though.

closed 2016-01-30 by joolswills