RetroPie/RetroPie-Setup #1219: Arcade system for all arcade emulators

HerbFargus, 2016-01-30:
I think it would eliminate some confusion to have an arcade rom folder for all the arcade systems as the user has the choice to select which emulator each rom uses. Would also consolidate a potential 4 systems depending on how many romset versions the user has.

joolswills, 2016-01-30:
I agree, but I would like to keep the existing folders still for those that want separate romsets etc.

HerbFargus, 2016-01-30:
Of course. I meant consolidation for those who would rather have them all in the same folder (just so 3 mame logos don't show up in emulationstation) we definitely should leave what is there currently for those who only have one romset.

joolswills, 2016-01-30:
Yep. sounds good :)

HerbFargus, 2016-01-31:
Resolved as of https://github.com/RetroPie/RetroPie-Setup/pull/1221

closed 2016-01-31 by HerbFargus