RetroPie/RetroPie-Setup #1224: Add ResidualVM

timfel, 2016-01-31:
May need to look at performance, but having ResidualVM available next to ScummVM may be nice.

joolswills, 2016-01-31:
Do you know if there is a working port currently ? I see a thread talking about there is GLES support for the openpandora, but nothing to say it is working/tested on the rpi.

timfel, 2016-01-31:
I found a reddit thread where people were discussing the performance on
RPi1 vs 2, but haven't tried myself, yet.

joolswills, 2016-01-31:
https://www.reddit.com/r/raspberry_pi/comments/2ys1l2/grim_fandango_on_the_pi/

joolswills, 2016-01-31:
Looks like they compiled it for X with software opengl, so it would be very slow. If there is an opengles backend, that would be the route.

timfel, 2016-02-01:
There is, they have a working android port. I just compiled ResidualVM straight on the RetroPie binary image from git, and it compiles fine with opengl renderer. Even running through X it's smooth. So I guess, I just copy the scummvm bits, adjust for residual, and wrap that into a PR?

joolswills, 2016-02-01:
I guess it must be using gles then. nice.  scummvm module should be fine to work off, but worth looking at the others to see how things are done. Sorry no dev documentation (yet) ;-) thanks.

if running under X you should check out the minecraft module as that starts up a minimal window manager etc. Did you try building /running without X - it would be preferable to do that (for the rpi )

joolswills, 2016-03-03:
This is included as an experimental module now.

closed 2016-03-03 by joolswills