RetroPie/RetroPie-Setup #1225: Retropie restarts after at emulations

istreva, 2016-01-31:
Hello guys, I've just installed my retropie a couple months back before 3.0 came out, but since i had done so I ran into several issues like sound and controllers configurations which reading i resolved.

_My main issue and still running is that my raspberry pi restarts at some point once I'm in retro pie and some times I'm able to run a few games, or even change configurations but I'm pondering what might be the issue,** I've changed the usb/mini usb connection (to the TV) and used a traveler charger of DC5V - 11V, **but the problem still persist, although I also have Openelec with Kodi on another SD card which at some point also restarts._

I'm wondering if anyone had these issues before or heard of anything similar that can be the solution to my problem.

thanks in advance for the responses.

joolswills, 2016-01-31:
Please use the forum for support https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

HerbFargus, 2016-02-20:
I don't believe this issue is due to retropie as others have not reported anything similar. Is best addressed on forum as above.

closed 2016-02-20 by HerbFargus