RetroPie/RetroPie-Setup #1226: input_shader_prev_axis input_shader_next_axis - Missing

nemo1966, 2016-01-31:
It seems input_shader_prev_axis and input_shader_next_axis are not supported or broken in the latest release.

joolswills, 2016-01-31:
does it work if you use other bindings ? Also have you tried building RetroArch from source ? This will need to be reported upstream to RetroArch if it confirmed an issue

nemo1966, 2016-01-31:
The save state slots and load slots Hotkeys work fine but not the shaders. I have reinstalled from Binaries but not the source.

nemo1966, 2016-01-31:
Below is my config - only the shaders NEXT and PREV will not work.

input_device = "raphnet.net (S)NES/Atari_USB"
input_driver = "udev"
input_l2_btn = "6"
input_start_btn = "3"
input_exit_emulator_btn = "3"
input_up_axis = "-1"
input_a_btn = "4"
input_b_btn = "0"
input_reset_btn = "0"
input_down_axis = "+1"
input_r2_btn = "7"
input_right_axis = "+0"
input_state_slot_increase_axis = "+0"
input_x_btn = "5"
input_menu_toggle_btn = "5"
input_select_btn = "2"
input_enable_hotkey_btn = "2"
input_y_btn = "1"
input_left_axis = "-0"
input_state_slot_decrease_axis = "-0"

# ======================================================================
# Config : raphnet.net (S)NES/Atari_USB : Hotkeys
# [Select] + [Start]      : Leave the game
# [Select] + [R]          : Save state
# [Select] + [L]          : Load state
# [Select] +           : Reset
# [Select] + [DPad-LEFT]  : Previous shader
# [Select] + [DPad-RIGHT] : Next shader
# [Select] + [DPad-DOWN]  : Decrease state slot
# [Select] + [DPad-UP]    : Increase state slot
# ======================================================================

input_save_state_btn = "7"
input_load_state_btn = "6"
input_reset_btn = "1"
input_shader_prev_axis = "-1"
input_shader_next_axis = "+1"

joolswills, 2016-01-31:
If you assign them to another input do they work (disable load/save and assign those buttons to the shader prev/next), or they only don't work when assigned to the axis ?

Please post the output of `/opt/retropie/emulators/retroarch/bin/retroarch --version`

nemo1966, 2016-01-31:
RetroArch: Frontend for libretro -- v1.3.0 -- 80bafb1 --
Compiler: GCC (4.9.2) 32-bitBuilt: Jan 17 2016

I don't know how to assign the shader to buttons other than the axis as the comand is "input_shader_prev_axis" <-- denotes using an axis (?)

I tried switching axis to -0 and +0 (the one that worked for the State stuff) and nothing. The state stuff worked on the -1 +1 axis as well.

joolswills, 2016-01-31:
Try building from source first, as that will build the latest code - it may already be fixed. if not, it will need to report it upstream to https://github.com/libretro/RetroArch (I can try and reproduce it and bisect when it broke, but I don't have a lot of free time for that right now).

gizmo98, 2016-01-31:
@nemo1966 
I don't think your settings are wrong. Open `/opt/retropie/emulators/retroarch/shaders`, take a look in all subfolders and copy interesting shaders into `/opt/retropie/emulators/retroarch/shaders`.

joolswills, 2016-01-31:
aha. it's a path issue ?

joolswills, 2016-01-31:
oh, the shader next/prev only handles shaders in the top level, and the structure was changed recently

gizmo98, 2016-01-31:
No. "I can imagine" (not tested) retroarch does only select shaders in the root shader path. If you use common-shaders repo all shaders are in subdirs now ;-) .

gizmo98, 2016-01-31:
Yeah. But we use the same structure as the libretro shader repo. So they should have the same problem.  In the past only the limited number of shaders in the top level could be used. But there are over 100 in subdirs which could be not used. Retroarch should look in subdirs as well.

joolswills, 2016-01-31:
gotcha. So technically something that needs fixing upstream still - maybe we can symlink some of the most popular shaders in the root as a workaround or something (just a few sensible ones that work well on the rpi)

joolswills, 2016-01-31:
(if it works with symlinks)

gizmo98, 2016-01-31:
All shaders which were toplevel in the past can be found under  `/opt/retropie/emulators/retroarch/shaders/retropie`. Just copy or symlink them into /opt/retropie/emulators/retroarch/shaders. But it's really just a small selection of all possible shaders.

joolswills, 2016-01-31:
yeh I think we should perhaps just choose some of the best, and symlink those to the subfolders perhaps, not necessarily just some of the retropie ones (many of them should be removed imho - floob did some testing on another ticket)

nemo1966, 2016-01-31:
OK I copied file from the crt folder to the shaders folders and pre selected one of those to start - I can confirm that now the shaders PREV and NEXT buttons now work. However I must be doing something else wrong as the shader does not change. I see the path change at the botton of the screen, but the actual shader does not. I even selected a shader that I know worked before.

gizmo98, 2016-01-31:
Just copy all and everything from the retropie folder into top level. This is the only difference to older versions.

nemo1966, 2016-01-31:
Got it working - thanks. I wasn't copying the shaders folder.  Thank you for the help!

gizmo98, 2016-01-31:
I will open a retroarch issue. Retroarch should take a look into subirs. I added the retropie folder so all shaders are sorted. If you take a look into your current folder it looks a little bit messed up. ;-)

nemo1966, 2016-01-31:
Great thanks - good to know something will be looked into!
Thanks again

joolswills, 2016-01-31:
Should RetroArch not be interested in the change (they may say for example just to copy the shaders you want to the folder pointed at by the config), I recommend we just symlink to copy shaders elsewhere, perhaps to the configuration area, so users can manage the list themselves.

joolswills, 2016-02-15:
I would like to sort this issue out as there are a few threads about it. My current thinking is to not use the entire repository of shaders for the rpi, and instead have a smaller selection in a flat directory structure. But I'm open to alternative suggestions.

gizmo98, 2016-02-15:
I will open up a branch for rpi and put all usable shaders there. Master branch should be still a ported copy of libretro common shaders. If you have a faster device (Intel NUC for example) you can use all other shaders.

joolswills, 2016-02-15:
sounds good - floob made a list on this issue tracker of ones which worked - we should def include the recent rpi crt one of course - also I think some even though they work do not need to be included - eg some of the scaling ones etc or the ones that make the screen unviewable :)

gizmo98, 2016-02-17:
I'm blind. Where can i find floobs list?

HerbFargus, 2016-02-17:
Should be here I think:

https://github.com/RetroPie/RetroPie-Setup/issues/933

gizmo98, 2016-02-20:
Ok. This can be closed now.

closed 2016-02-20 by gizmo98