RetroPie/RetroPie-Setup #1228: Text prompt for emulator options not always showing before play

vretro, 2016-01-31:
Using the latest RetroPie image on Pi2, on occasion, the text prompt alerting the user that they can press a key to configure emulator options does not always show just before a game starts.

The game will start when this issue is present, however tapping a button to enter the menu (in the absence of the prompt) seems to leave the user with a black screen and the game does not boot.

This can occur after exiting a game, where the prompt would have been displayed, then from the Emulation Station menu, selecting another game, where the prompt would then be missing, with the potential for leaving the user with a blank screen if anything is pressed before the game loads.

I tried a full binary reinstall to fix, but the same issue remains. Also tried Raspian software update.

joolswills, 2016-01-31:
The console blanker is just activating in the jessie image - press a key on a keyboard and it will come back. Actually a Debian Jessie bug - see this workaround - http://blog.petrockblock.com/forums/topic/2-issues-on-3-4-that-werent-there-on-3-3/#post-115077

joolswills, 2016-01-31:
>  the console blanker is still active. https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=771161
> Try adding “consoleblank=0” to the end of the existing line in /boot/cmdline.txt
> or alternatively (safer as less likely to break booting by messing with cmdline.txt) add “setterm -blank 0” to /etc/rc.local (above the exit 0 line)

vretro, 2016-02-01:
Thank you Jools.

It is a really unfortunate show stopper bug, as it recurs and has left me with a blank screen frequently with no game or obvious way to return to Emulation Station.

I have been rebooting via terminal on another device as a "fix".

vretro, 2016-02-03:
Just a quick followup:

I added consoleblank=0” to the end of the existing line in /boot/cmdline.txt – and the issue still happens. Will try your alternative suggestion next.

joolswills, 2016-02-03:
use the other fix - it's safer. please post your cmdline.txt though as I suspect you may have added it wrong (unless the single quote above is a typo)

vretro, 2016-02-04:
cmdline.txt contains:
consoleblank=0

It was a typo on the GitHub issue only as I had quoted your previous post and reworked the text before posting.

Will go with your second suggestion as directed, add the following above exit 0, in /etc/rc.local:
setterm -blank 0
Thanks

joolswills, 2016-02-04:
you have edited the wrong file - it should contain other stuff the file is located at /boot/cmdline.txt - it doe work, but the other fix is safer so use that.

vretro, 2016-03-15:
After booting a fresh install of Raspian then installing RetroPie, the issue of the console blanker still being active is present. I have applied the suggested fix in rc.local and was wondering if this fix should be installed as part of a standard RetroPie installation?

HerbFargus, 2016-03-15:
As far as I understand it that is added to the image we provide. For those installing on raspbian manually have to make those changes themselves manually.

joolswills, 2016-03-15:
yep - it is added to /boot/cmdline.txt on the image as detailed above.

joolswills, 2016-03-15:
This can be closed now.

closed 2016-03-15 by joolswills