RetroPie/RetroPie-Setup #1229: Super Mario War Netplay

HerbFargus, 2016-01-31:
Opening a ticket just so I have a reference.

Currently testing Super Mario War Netplay and have a working module, still in development. There is also a build with SDL2 that may be worth looking into https://github.com/mmatyas/supermariowar/wiki/Raspberry-Pi-2-Compiling-Guide#sdl2-script

```
#!/usr/bin/env bash

# This file is part of The RetroPie Project
# 
# The RetroPie Project is the legal property of its developers, whose names are
# too numerous to list here. Please refer to the COPYRIGHT.md file distributed with this source.
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
#

rp_module_id="smw-netplay"
rp_module_desc="Super Mario War (Netplay Build)"
rp_module_menus="4+"

function depends_smw-netplay() {
    getDepends libsdl1.2-dev libsdl-mixer1.2-dev libsdl-image1.2-dev
}

function sources_smw-netplay() {
    gitPullOrClone "$md_build" https://github.com/mmatyas/supermariowar.git
    git submodule update --init
}

function build_smw-netplay() {
    cd "$md_build"
    unzip data.zip
    mkdir Build && cd Build
    cmake ..
    make clean
    make -j 4

    md_ret_require="$md_build/Build/Binaries/Release"
}

function install_smw-netplay() {
    md_ret_files=(
        '/Build/Binaries/Release/smw'
        '/Build/Binaries/Release/smw-server'
        '/Build/Binaries/Release/smw-worldedit'
        '/Build/Binaries/Release/smw-leveledit'
        '/data'
    )
}

function configure_smw-netplay() {
    mkRomDir "ports"

    addPort "$md_id" "smw-netplay" "Super Mario War Netplay" "$md_inst/smw $md_inst/data"
}
```

Ronoh55, 2016-02-02:
Module for SDL2 compiled fine but wouldn't launch (3.4/Jessie) 

This module does seem to be causing my black screen on Retropie setup launch, but as we said on irc there seems to be no reason it is. All other settings are default.

HerbFargus, 2016-04-04:
As this is more or less long term before considered stable and already included in @zerojay extra modules I will close this for the time being and revisit it once the netplay functions are stable.

closed 2016-04-04 by HerbFargus