RetroPie/RetroPie-Setup #1231: look into using xboxdrv for joystick to keyboard for kodi

joolswills, 2016-02-01:
Should be able to use xboxdrv to map joystick event device to keyboard, to workaround using a kodi without joystick support built in (the raspbian shipped one).

zerojay, 2016-02-09:
I started looking into this myself and I think one of the things that would really help out our implementation is how we handle xboxdrv. Right now, we write our options directly into rc.local. I think it would be better if we either used xboxdrv as a daemon (using /etc/init.d/xboxdrv start to run) or if we wrote out all our xboxdrv options out to a separate file (/etc/xboxdrv.sh?) that we would run from rc.local. This would make things a lot easier for us to kill our current xboxdrv when launching Kodi so we can start Kodi up with its own xboxdrv options and then relaunch after we quit Kodi. Anytime we'll want to reinitialize xboxdrv, we can just run that script instead.

Doing some testing on this now.

joolswills, 2016-02-09:
having an init.d script would be better if we handle the configuration properly - using the -c config file parameter (there is an example in the wiki that is not an improvement really imho).

joolswills, 2016-04-09:
At least for the RPI this is no longer needed as the pipplware repo we use has joypad support - but we still need to generate the  config automagically, so I shall leave this open for that.

joolswills, 2016-04-09:
actually no I wont - it makes this thread confusing.

closed 2016-04-09 by joolswills