RetroPie/RetroPie-Setup #1233: Retroarch: Next attempt to enable save on exit

gizmo98, 2016-02-02:
I like RGUI, but its annoying it cannot be used to setup things persistent. Newbies will be confused. Experienced users need our shell scripts or an editor to set things. All efforts enabling save on exit were disastrous by now. Replacing appendconfig or using #include does not help. We could also disable RGUI at all.

I've taken the liberty to write a startup script which makes a copy of our current system config, enables save on exit, starts retroarch with this copy and writes after exit all proved to be save settings to system and global config file.
https://github.com/RetroPie/RetroPie-Setup/compare/master...gizmo98:retroarch-save-on-exit?expand=1

Things like input/audio/video "driver" stuff will be written to configs/all. All shader/overlay/audio stuff will be written to configs/$system.

Don't know if this a viable way. It seems to work. What do you all mean?

HerbFargus, 2016-02-02:
I like the idea, I don't have time to test anything right now but managing settings through the rgui seems a lot more intuitive than needing to always set configs manually. 

If there is some way to integrate it with runcommand so that it matches the settings per rom or per system that would be ideal.

gizmo98, 2016-02-03:
Integrate settings per rom should be easy. If $rom.cfg exists store settings in this file. Else use system.cfg file.

objectchris, 2016-02-03:
I like it! personally one of the first things I do with a retropie install is turn save after exit on. there's just too many things I need to change and the RGUI is the best way to do that IMO.

joolswills, 2016-02-04:
Yeh the idea is nice. There might be cases where it saves something where someone doesn't want it  but generally I think it covers what most people would want - maybe we should leave it off by default, and get some testing done first though as it's a fairly big functionality change.

gizmo98, 2016-02-06:
Ok. I disabled it by default. Do you have any tip how to speed up ini read and write operations? It takes 2-3s to store all settings. I know python's configparser can do this in memory (read whole file --> edit everything --> write file back). I had another problem using arguments with whitespaces like rom name and rom config. I worked around it but there could be a better solution.

joolswills, 2016-02-06:
for args just process the array of args you get and pass that on

eg the content of "$@" or so but the ini / read write will re-read the whole file each time, so there should be ways to speed it up.

joolswills, 2016-02-06:
and of course there is the option of using python should shell scripting prove too slow.

joolswills, 2016-02-06:
just did some experimentation - here's a way to speed it up.

You generate some expressions sed the keys you want from the first file, replacing them with an expression to replace them in the second file.

eg

```
#!/bin/bash

source "scriptmodules/inifuncs.sh"

# system specific settings
local_vars=(
    video_scale
    video_vsync
    #video_force_srgb_disable
    video_hard_sync
    video_hard_sync_frames
    video_frame_delay
    #video_black_frame_insertion
    video_threaded
    #video_shared_context
    video_smooth
    video_force_aspect
    video_scale_integer
    video_aspect_ratio
    video_aspect_ratio_auto
    video_crop_overscan
    video_shader
    video_shader_enable
    #video_refresh_rate
    video_allow_rotate
    video_rotation
    audio_sync
    #audio_latency
    audio_rate_control
    #audio_rate_control_delta
    #audio_max_timing_skew
    input_player1_analog_dpad_mode
    input_player2_analog_dpad_mode
    input_player3_analog_dpad_mode
    input_player4_analog_dpad_mode
    input_player5_analog_dpad_mode
    input_player6_analog_dpad_mode
    input_player7_analog_dpad_mode
    input_player8_analog_dpad_mode
    input_overlay_enable
    #input_overlay_hide_in_menu
    input_overlay
    input_overlay_opacity
    input_overlay_scale
    custom_viewport_width
    custom_viewport_height
    custom_viewport_x
    custom_viewport_y
    fps_show
    rewind_enable
)

iniConfig " = " ""
re=("-rn")
for var in "${local_vars[@]}"; do
    re+=("-e" "s|^ ?$var ?= ?\"(.+)\".*$|s/^ ?$var ?=.*/$var = \"\1\"/|p")
done
nre=()
while read -r line; do
    echo $line
    nre+=("-e" "$line")
done < <(sed "${re[@]}" /opt/retropie/configs/all/retroarch-tmp.cfg)
sed -r "${nre[@]}" /opt/retropie/configs/all/retroarch.cfg
```

`re+=("-e" "s|^ ?$var ?= ?\"(.+)\".*$|s/^ ?$var ?=.*/$var = \"\1\"/|p")` i not the most readable line of code ever, but it works in my testing, and is significantly faster 0.35s vs 2.47s for doing it one value at a time.

joolswills, 2016-02-06:
updated the script slightly

joolswills, 2016-02-06:
as retroarch always saves out it's configs with the same spacing etc, I have simplified the regular expression - more readable now :)

joolswills, 2016-02-06:
this is not a complete solution actually, as it will only replace existing keys. but I'm sure it could be improved to replace existing keys, and then add any new ones.

joolswills, 2016-02-06:
ok so a bit more tweaking - this "should" append new lines, and replace existing ones. Takes about 0.45s to run on my rpi2

```
for var in "${local_vars[@]}"; do
    grep_re+=("-e" "^[# ]*$var")
done

matches=($(grep -E -o "${grep_re[@]}" /opt/retropie/configs/all/retroarch.cfg | tr -d "# "))

for var in "${local_vars[@]}"; do
    if [[ " ${matches[*]} "  =~ " $var " ]]; then
        re+=("-e" "s|^$var = \"(.+)\"|s/^[# ]*$var *=.*/$var = \"\1\"/|p")
    else
        re+=("-e" "s|^$var = \"(.+)\"|$ a $var = \"\1\"|p")
    fi
done

while read -r line; do
    nre+=("-e" "$line")
done < <(sed -rn "${re[@]}" /opt/retropie/configs/all/retroarch-tmp.cfg)

sed -r "${nre[@]}" /opt/retropie/configs/all/retroarch.cfg
```

gizmo98, 2016-02-07:
Wow. This is really nice stuff.

joolswills, 2016-02-07:
Thanks - well I'm not sure it's too pretty, but it works :)

dankcushions, 2016-03-04:
One thing to be aware of, if you globally save_on_exit set to true, and you're running a per-game override romname.ext.cfg (eg, set up via the runcommand), and you exit said game, then all the override settings will be written to your /core/retroarch.cfg, which isn't normally what you'd want.

I've gotten around this by adding this at the end of all my per-game configs:
````
# Never save-on-exit after an override config
# or the override will make into the core config.
config_save_on_exit = false
````
This means the override can be loaded without affecting anything else.

Perhaps there's a better way of ignore override config options with your script?

meleu, 2016-07-14:
[Just registering here what I've posted in the forum]
I have some near friends that are used to use RetroArch and RGUI on Android, and frequently they get confused with RetroPie's way of dealing with retroarch.cfg files. And we see here several threads with people facing the same kind of problem.

I was thinking in a way to make the RGUI fans happy: Maybe an entry in every .../SYSTEM/emulators.cfg with an option loading the same alternative retroarch.cfg with config_save_on_exit = true. This way the user can choose this "emulator" with runcommand.

Pros:

the obvious: make the RGUI fans happy (those people that use RetroArch on other platforms).
easy to revert to the default RetroPie way of manage retroarch.cfg files.
doesn't mess the specific system's retroarch.cfg.
Cons:

the alternative retroarch.cfg file will be dificult to manage manually. But hey! this file is not intended to be edited manually!
Can you, experienced guys, see other cons (or pros) in this approach, that I'm not seeing?

meleu, 2016-08-13:
What do you guys think about the solution proposed here:
https://retropie.org.uk/forum/topic/3285/i-think-we-finally-have-a-solution-for-rgui-fans