RetroPie/RetroPie-Setup #1234: gamecon_gpio_rpi Module is failing to launch on boot with RetroPie 3.4

w3ace, 2016-02-03:
I noticed this issue 2 weeks ago while testing Jessie BETA on Pi Zeros. I switched to Wheezy, but now am seeing other cases of this with the release of 3.4 - [See Instructable ](http://www.instructables.com/id/RetroPie-Inside-an-NES-Controller/)

1. The gamecon_gpio_rpi module will not load on boot.

`modprobe gamecon_gpio_rpi map=0,0,2,0,0,0` will work fine, but adding gamecon_gpio_rpi map=0,0,2,0,0,0 to /etc/modules ends up having the kernel modules fail to load.

auspex-nl, 2016-02-04:
I ran into the same problem (after some errors, got the kernel module compiled and loadable with modprobe)

This works for me:

Create a file `/etc/modprobe.d/gamecon.conf` with just this single line (remember to replace map with your config:
`options gamecon_gpio_rpi map=0,0,2,0,0,0`

You could confirm it's working by running `modprobe gamecon_gpio_rpi`

Now in `/etc/modules` add just the module name and omit the map parameter:
`gamecon_gpio_rpi`

Now your module should load fine on boot.

joolswills, 2016-02-04:
module options should not have been in /etc/modules even on wheezy - if it worked, it was luck or some legacy functionality. Feel free to add information to our wiki about this.

joolswills, 2016-02-04:
Yeh I think that is just a legacy way. So no RetroPie bug - btw it's worth pointing our this has very little to do with RetroPie - this is all related to the underlying OS. RetroPie is just a bunch of software that sits on top of Raspbian.

HerbFargus, 2016-02-04:
Here's the link for convenience:

https://github.com/retropie/RetroPie-Setup/wiki/GPIO-Modules

joolswills, 2016-02-05:
I hadn't realised we had code in retropie-setup that also did this, so that needs fixing. I never wrote this code and it's not been changed a lot (apart from some cleanups). I'll keep this open as a reminder that it needs fixing in gamecondriver.sh

HerbFargus, 2016-02-05:
Also a post for reference:

http://blog.petrockblock.com/forums/topic/game-on-setup-issue/

joolswills, 2016-02-11:
This should be fixed now

commit 7840a73c (2016-02-11): put gamecon_gpio_rpi options in /etc/modprobe.d/gamecon.conf and remove from /etc/modules if there
use printMsgs instead of echo
fixes #1234
closed 2016-02-11 by joolswills