RetroPie/RetroPie-Setup #1235: mame2003-libreto coin joypad not always working

vretro, 2016-02-03:
Using an iBuffalo joypad, the Select button which usually inserts coins for player1/2 doesn't always function and I'm not sure if it is a per game issue.

Issue experienced using 1 joypad attached.

joolswills, 2016-02-03:
You need to change the hotkey enable button 

https://github.com/RetroPie/RetroPie-Setup/wiki/FAQ

joolswills, 2016-02-03:
Please use the forum for support / help first - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

vretro, 2016-02-04:
Sorry to bother with this issue.

I was aware of the work arounds, just viewed it as a bug. Upon further reading, I see that you have worked on resolving this issue with hotkey conflicts in RetroArch and created a patch #1072

Is your patch something that will get rolled into RetroPie in future as you decide, or is it dependant on the Libretro team including your patch?

closed 2016-02-04 by HerbFargus

joolswills, 2016-02-04:
the patch doesn't work and it's not a fix. it's not a bug really either - read through the RetroArch thread for more info.

HerbFargus, 2016-02-04:
It's already logged as an issue here:

https://github.com/RetroPie/RetroPie-Setup/issues/1072

There have been changes since the patch and it likely doesn't work. May be a while before its sorted as retroarch was designed to use an external hotkey that is not also used for a game function (such as coin inserts)

objectchris, 2016-02-04:
IMO having even a faulty patch (ie, one that ignores all hotkey + [button] blocking rather than just allowing the hotkey through), would currently have less annoying side effects than the current situation. It's really frustrating that this is still a problem - it makes a keyboard-less system impossible for me, unless i apply my manual patch every time i want a new version :(

joolswills, 2016-02-04:
I may consider turning off the hotkey blocking or adding an option, but there was a lot of code changes upstream recently and it will be a pain maintaining customisations. I think impossible is the wrong term - reassign the button, get a controller with more buttons etc. It's not really RetroPie's fault either. Your best bet would be to try and get some option to disable hotkey blocking added upstream - but if they won't then you will have to workaround it for now.

It's as much work for us to maintain a custom patch as it is for you btw.

objectchris, 2016-02-04:
@joolswills I felt like our fixes (to allow the select key to be passed to the game UNTIL an additional paired key was pressed at the same time) would be accepted by retroarch (they said as much), if only we could get them working :) my C guesswork is only good enough to disable the hotkey blocking completely, and we gatekeep retroarch from the amount of users getting this problem so i don't think they appreciate how many people it affects, so i feel like it will remain a wontfix.

i say impossible because i like to use a controller appropriate to the systems I'm using. right now you'd have to have a controller with more buttons than the PS1 controller (the most button-heavy system in retropie?), which leaves what? re-assigning just means the problem is now moved to some other button.

i know, should just learn C and fix it myself :)

HerbFargus, 2016-02-04:
If you can get it working I would applaud you. I don't like the disabling patch because on systems other than mame it affects gameplay when setting a savestates and would be more inconvenient than changing the retroarch.cfg for one or two systems. 

I agree its something that needs to be fixed but you're right in that with opensource if something isn't working everyone has the opportunity to fix it :)

joolswills, 2016-02-04:
Send the users over to RetroArch - one of the frustrating parts of working on RetroPie is that we get bug reports for everything from the Kernel all the way up - but in fact we are just a set of packages running on top of Raspbian. We cannot be responsible for everything and I don't think we should have to be.

objectchris, 2016-02-04:
@HerbFargus the ironic thing is that the way retroarch works means that it often DOESN'T block the key anyway! it always lets the first input frame through before it starts blocking so for example in legend of zelda on the NES, if you do a [select] (assuming select is your hotkey) + [save button] it will save with game paused (select is the pause button).

when it works (and therefore when the blocking causes the hotkey to not function in-game) is for games that require the buttons pressed for multiple frames, or held. this is why the insert coin works in SOME mame/fba games regardless.

@joolswills i agree. right now i follow this project and i see us redirecting users to the wiki with the workarounds. I get that this is the quickest solution for the users but maybe we should refer the call in https://github.com/libretro/RetroArch/issues/2230 also so there's a central record of the numbers. i think if you added together the issues in git + forums it would be 50+ by now.

objectchris, 2016-02-05:
@HerbFargus ironically retroarch doesn't block properly at the moment either! it always lets the first input frame through so for games where the hotkey still functions in-game (ie, the game is not looking for more than one frame of input to trigger its function), they will register regardless. eg in the legend of zelda (NES) it will always start up save states in 'paused' mode, if you save via the default select+L (or R, I forget). 

in my experience most console games will not have the hotkey blocked (so your hotkey functions fine in games, but you'll get other issues), but most mame/arcade games will (so you get the insert coin issue). i figure this is because coin mechanisms normally want more than one frame of signal to register, to stop false positives?

@joolswills agreed. perhaps instead of just giving users a link to the wiki page of workarounds we should also add the occurrence to https://github.com/libretro/RetroArch/issues/2230 so it's visible - i reckon if we totalled up all the users mentioning this on the forum, git, etc, it would be pretty high!