RetroPie/RetroPie-Setup #1237: [request] N++ game

vrtlspd, 2016-02-04:
Would it be possible to add this game? I remember playing this on xbox360 with 3 other friends and it was very funny. A well spent afternoon :)

I hadn't much time to try to make this work but if you go to the download page you have an swf file, so its just a matter finding a way to make it run.
I stumbled across this http://www.gnu.org/software/gnash/download.html but I haven't tried it yet.

http://www.thewayoftheninja.org/n.html

would this be possible? would it run at decent framerate?

joolswills, 2016-02-04:
gnash is available in raspbian jessie, so you can install it and try it out perhaps ? It will probably be horribly slow, and it will require X etc. I wouldn't think it would be a candidate for adding to RetroPie really.

vrtlspd, 2016-02-04:
thank you for the recommendations. when I have time i'll try it

joolswills, 2016-02-04:
Let us know how you get on. thanks.

HerbFargus, 2016-02-04:
Just for reference I've used gnash (in the lxde environment) and all the flash games i tried ran horribly slow (on my pi 2 at that) .  I tried it with super Mario crossover, barely made it past 10 frames.

zerojay, 2016-02-07:
Also, even on Windows with various Flash game running tools, N+ would often run in a tiny tiny resolution that would make it essentially unplayable.

vrtlspd, 2016-02-08:
well, I couldn't get the swf file on the official page to run. I opened it with gnash but nothing happens... I tried an older version from some other website.
The game is playable but it lags sometimes. I would say it runs around 20fps.
One thing I dont understand is, raspbian has a cpu usage graphic but it never passed the 25% mark. Is raspbian only running in one cpu?

some good news:
http://www.pcgamer.com/n-is-coming-to-steam/?utm_content=buffercccff&utm_medium=social&utm_source=facebook&utm_campaign=buffer_pcgfb

joolswills, 2016-02-08:
https://askleo.com/why-is-only-one-core-on-my-multi-core-processor-being-maxed-out/

zerojay, 2016-02-09:
I would have suggested N+ on PSP but it runs extremely slowly - a surprise considering it looks like it's just a pure 2D game.

HerbFargus, 2016-02-17:
I don't see this as something viable for the pi if based in flash.

closed 2016-02-17 by HerbFargus