RetroPie/RetroPie-Setup #1240: lr-pcsx-rearmed not working with xboxdrv

vrtlspd, 2016-02-05:
since I enabled xboxdrv the psx emulator keeps crashing right at the startup. It doesn't even load any game.

do you need logs or something?

joolswills, 2016-02-11:
Please make sure you are running the latest retropie (v3.5) and that you have the correct bios installed for the emulator. I wouldn't think the xboxdrv could cause it to crash, but retroarch will log to /tmp/runcommand.cfg which you can sftp off after a crash.

HerbFargus, 2016-02-22:
It runs just fine for me. I tested with ctr with my xbox 360 controller. perhaps its a bios issue or issues with your game files.

joolswills, 2016-02-22:
thanks. @vrtlspd please use the forum for further help with this - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2016-02-22 by joolswills