RetroPie/RetroPie-Setup #1242: Adding screenshot.sh

Omnija, 2016-02-07:
Guide/Tutorial
-------------------
http://pastebin.com/uPYvF8vF

Screenshot of  Proof of concept
--------------------
http://i.imgur.com/06griWS.png

HerbFargus, 2016-02-07:
Heres a working Module:

create a file called `screenshot.sh` in  `/home/pi/RetroPie-Setup/scriptmodules/supplementary`

and then it can be installed from the experimental section of the retropie setup script. 

```
#!/usr/bin/env bash

# This file is part of The RetroPie Project
# 
# The RetroPie Project is the legal property of its developers, whose names are
# too numerous to list here. Please refer to the COPYRIGHT.md file distributed with this source.
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
#

rp_module_id="screenshot"
rp_module_desc="Universal Screenshot with Raspi2PNG"
rp_module_menus="4+"

function depends_screenshot() {
    getDepends libpng12-dev
}

function sources_screenshot() {
    gitPullOrClone "$md_build" https://github.com/AndrewFromMelbourne/raspi2png.git
}

function build_screenshot() {
    cd "$md_build"
    make clean
    make
    md_ret_require="$md_build"
}

function install_screenshot() {
    md_ret_files=(
        'raspi2png'
        'LICENSE'
        'README.md'
    )
}

function configure_screenshot() {
    mkUserDir "$datadir/screenshots"

    # Create script to take screenshot over ssh
    cat > "$datadir/screenshots/screenshot.sh" << _EOF_
#!/bin/bash
$rootdir/supplementary/screenshot/raspi2png -p $datadir/screenshots/$(date +%m%d%Y_%H%M%S).png
_EOF_

    chown $user:$user "$datadir/screenshots/screenshot.sh"
    chmod +x "$datadir/screenshots/screenshot.sh"

    printMsgs "dialog" "To take a screenshot use $datadir/screenshots/screenshot.sh over SSH."
}
```

To add it to samba shares needs to be another line here:

https://github.com/RetroPie/RetroPie-Setup/blob/59e31b525bf25d80e1ce4f365870c48f1889d10b/scriptmodules/supplementary/samba.sh#L53

HerbFargus, 2016-03-08:
Also another thought for this, While raspi2png may be helpful for troubleshooting, most errors are logged to tmp directories that are easily accessible so it may actually be better utilised as a function for users to build their own artwork for EmulationStation (particularly considering the recent instability of thegamesdb). 

Sselph's scraper provides the ability to link roms to images and produce gamelists in place of the default scraper. Would need some extra coding in runcommand and the scraper module to implement.

see related: https://github.com/RetroPie/RetroPie-Setup/issues/1358

Might also be worth integrating some optimisation for the screenshots with something like pngcrush or pngquant see more info on png optimistation [here](http://pointlessramblings.com/posts/pngquant_vs_pngcrush_vs_optipng_vs_pngnq/)

joolswills, 2016-03-08:
yeh that's doable.

joolswills, 2016-03-08:
(albeit lower priority than some other things ongoing)

HerbFargus, 2016-03-09:
### Screenshot Comparison Between RetroArch vs Raspi2png

#### RetroArch (4:3) (Filesize: 16 KB)

[image]

#### Raspi2png (4:3) (disable overscan not set)  (Filesize: 449 KB)
[image]

(with overscan disabled) (Filesize: 482 KB)
[image]

#### Retroarch (16:9) (Filesize: 17 KB)
![retroarch-0309-014917]
(https://cloud.githubusercontent.com/assets/10035308/13622807/9c214810-e55e-11e5-9fdd-99db7a0e13cb.png)

#### Raspi2png (16:9) (Filesize: 492 KB)
[image]

### PNGQUANT OPTIMISATION
163 KB
[image]
175 KB
[image]
185 KB
[image]

(Original Files)
[Screenshot Comparison.zip](https://github.com/RetroPie/RetroPie-Setup/files/164313/Screenshot.Comparison.zip)

HerbFargus, 2016-06-19:
its in the retropie extra repo for the time being and maybe will address in the future if the quality and filesize can be resolved.

closed 2016-06-19 by HerbFargus