RetroPie/RetroPie-Setup #1244: RPI2 UAE4Arm setup script always compiles RPI1 version

windale, 2016-02-07:
Im on a Pi 2 and downloaded the Pi 2 version of Retropie. When updating UAE4Arm from Source it always compiles the Pi 1 version (which has Picasso/RTG memory disabled etc.) As a quick workaround, I changed part of the UAE4Arm setup script from this 

**if isPlatform "rpi"; then
make PLATFORM=rpi1 CPU_FLAGS=""
else
make PLATFORM=rpi2 CPU_FLAGS="-mfpu=neon"**

to this :-

**if isPlatform "rpi"; then
make PLATFORM=rpi2 CPU_FLAGS="-mfpu=neon"
else
make PLATFORM=rpi2 CPU_FLAGS="-mfpu=neon"**

and now it compiles the correct version. Can you please look into this so the Pi 2 version is correctly compiled when on a Pi 2 / Retropie for Pi 2 version. Thanks,

joolswills, 2016-02-07:
thanks for the heads up, yeh the code is wrong - but it should be "if isPlatform "rpi1"; then ill sort it. thanks.

joolswills, 2016-02-07:
fixed.

joolswills, 2016-02-07:
I have also updated the binaries.

closed 2016-02-07 by windale