RetroPie/RetroPie-Setup #1246: set default retroarch controller

fopina, 2016-02-07:
I have used in the past RetroPie 2.3, the official SD card image and everything worked perfectly out of the box with my gamepads (thrustmaster t mini wireless).

Last week I installed Retropie on top of Jessie (RPi2) and the gamepads were not used by RetroArch.
They worked in ES, the yellow text with their IDs was shown when a ROM started but the only game conroller working was the usb keyboard...

I noticed that my keyboard was /dev/input/js0 (before I think it wasn't detected as joystick) and the gamepads were js1 and j2, so I decided to unplug my keyboard and suddenly the selected controller was one of the gamedpads.

I've tried changing udev rules to have the gamepads as js0 and js1 (and the keyboard as js2) but it still selects the keyboard as the controller.
How can I choose which controller controls the ROM?

I've been going through wiki and google results, but there seems to be no option...

joolswills, 2016-02-07:
change the order they are connected (top left usb = 0, bottom left =1, top right 2, bottom right 3)

For further help please use the forum 
https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

fopina, 2016-02-07:
Sorry!
I will try changing the USBs to check it out, in the meantime it seems to have also worked to change the indexes in retroarch.cfg (I decided to read the whole file trying to find something)

```
# If desired, it is possible to override which joypads are being used for user 1 through 8.
# First joypad available is 0.
input_player1_joypad_index = 1
input_player2_joypad_index = 2
```

Closed and I'll post in the forum if I need further help, thanks.

closed 2016-02-07 by fopina