RetroPie/RetroPie-Setup #1248: Upgraded to 3.5 from 3.4 image - EmulationStation not loading

smartroad, 2016-02-08:
Have upgraded from a 3.4 image to 3.5 and now ES starts with its splashscreen but then just goes white. Have I missed something?

HerbFargus, 2016-02-08:
What theme are you using?

https://github.com/retropie/RetroPie-Setup/wiki/themes#white-screen-of-death

joolswills, 2016-02-08:
sounds to me as though a non existing theme is being referenced or something - they can configure it or install the default theme or something - this is more suitable for a post on the forum rather than this issue tracker though as it isn't a bug https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

smartroad, 2016-02-08:
I have tried to kill emulationstation but there isn't anything listed. ES starts then bombs out. I just tried to launch it from command line and got:

lvl0: 	Error creating SDL window!
	Could not create GLES window surface
lvl0: 	Renderer failed to initialize!
lvl0: 	Window failed to initialize!

I also tried to compile it from source with no luck. It can't be the theme as the theme worked fine before (PI2 with 256MB spared for GFX) and still exists in the right directory. This worked all okay before I upgraded. Someone else also mentioned it on the blog page that they have a white screen after a binary install.

smartroad, 2016-02-08:
So far I have stopped emulationstation starting on boot (and stopped the splash screen loading). I performed a binary update to it from the script, checked which theme it was trying to load in the cfg file (simple-dark), checked that simple-dark is installed via the script (it is, but installed it again for good measure), quit out of the script, rebooted, logged back in and tried to run emulationstation and same thing, the ES logo appears and then a white screen. I can't even ctrl+alt to another terminal, i have to ssh into the machine to get it to do anything.

InsecureSpike, 2016-02-08:
how many emulators do you have running?

smartroad, 2016-02-08:
6 or 7 but like I say it was working fine before the update.

joolswills, 2016-02-09:
please post a log from after you get a white screen. Also please switch to the carbon theme for now.

joolswills, 2016-02-09:
please post a copy of /boot/config.txt also.

smartroad, 2016-02-09:
I edited the es_settings.cfg to carbon and started up ES. ES worked, so I changed (inside ES) back to simple-dark and it changed back without any issues. So all seems well, just manually changing the theme worked. Many thanks guys!

The question is though, why did it hickup in the first place? I have updated in the past with no issues like this. The theme was working okay and after a little manual poking it is okay now. What was it about updating that cause it? When I saw that RP had released 3.5 I SSH's into my machine and run the retropie_setup.sh - upgrading the script first and then doing a binary update. Could it be, as HerbFargus suggested, that ES was running at the time of the update and that upset it? Is it possible for the script to tell if it is being run through an external terminal to kill the ES process before upgrading it and then running it again after or just reboot the machine?

joolswills, 2016-02-09:
no - linux filesystems don't work like that, not sure what the problem was tbh, perhaps a theme path change or something (so it wasn't picking up the theme from the right location). Have you installed themes manually for example ?

smartroad, 2016-02-09:
I have only installed themes via the script but I do have several installed but that was while I was finding which theme worked best for me.

joolswills, 2016-02-09:
I don't know the cause then. Sorry. But glad it's solved.

smartroad, 2016-02-09:
Thanks for the help :grin:

closed 2016-02-09 by smartroad

smartroad, 2016-02-11:
Have started a thread in the forum http://blog.petrockblock.com/forums/topic/white-screen-on-boot-with-simple-dark-theme/ as it is still happening. If I boot using simple-dark ES whitescreens. If I boot with carbon, switch to simple-dark, restart only ES the theme loads fine. If I then reboot, whitescreen again.