RetroPie/RetroPie-Setup #1253: Default Arcade system theme to MAME theme to avoid blank screen.

zerojay, 2016-02-09:
I would have gone ahead and sent in a pull request however the generation of es_systems.cfg isn't exactly straightforward as far as I can tell. Appears that the change will either need to be made in scriptmodules/helpers.sh or scriptmodules/emulators/mame4all.sh. Here's the default settings for arcade in es_systems.cfg.

  <system>
    <name>arcade</name>
    <fullname>Arcade</fullname>
    <path>/home/pi/RetroPie/roms/arcade</path>
    <extension>.fba .zip .FBA .ZIP</extension>
    <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 _SYS_ arcade %ROM%</command>
    <platform>arcade</platform>
    <theme>arcade</theme>
    <directlaunch/>
  </system>

We'll want to change <theme>arcade</theme> to <theme>mame</theme> so that the arcade system does not have a plain white theme for no real reason.

HerbFargus, 2016-02-09:
The default carbon theme has an arcade theme so the only reason it would be white is if the theme someone is using doesn't have a theme for arcade yet. Might be more productive to add the arcade theme to the themes that are missing it.

zerojay, 2016-02-09:
Ah, it was the pixel theme that I was using at the time.

HerbFargus, 2016-02-09:
You can petition rookervik on the retropie irc to make an arcade theme for pixel or brush up on your pixel art skills and submit a PR. 

I'd just as well people change the theme to mame manually if the theme is missing arcade as arcade includes more than just mame.

InsecureSpike, 2016-02-09:
what is the irc name?

zerojay, 2016-02-09:
Rookervik is working on an arcade theme now for Pixel.

closed 2016-02-11 by joolswills

HerbFargus, 2016-02-15:
I have added it to the source https://github.com/RetroPie/es-theme-pixel/commit/884879f7621db25264faedd67d7d52b2e56d6ed3. For future reference for requests on theming, please open tickets on the theme's respective github repository (most should be a part of the retropie organisation now)