RetroPie/RetroPie-Setup #1254: [request] rpix86 start script

BenWlson, 2016-02-09:
When comparing +Start DOSBox.sh and +Start rpix86.sh, I can tell that the DOSBox script is built to handle other scripts as an input (in addition to bat/com/exe) that you can write to load specific games, and without any input, it just loads DOSBox in a default manner. Is there a way to revise the +Start rpix86.sh script to behave the same way? Right now it looks like it can only handle a bat/com/exe file as an input to apply as a parameter to the executable. In order to clean up ES, I removed everything but .sh files from the es_systems.cfg file and would like to be able to create scripts to load games in rpix86. I imagine the scripts for individual games would only work with either DOSBox or rpix86, but we should be able to set a default emulator for those scripts using the runcommand feature.

This feature would be nice as I occasionally find a program that runs better in rpix86.

joolswills, 2016-02-09:
Can you give me an example of a script you would want it to handle ?

joolswills, 2016-02-09:
the reason dosbox supports scripts is to handle custom configurations per game etc - so am interested in a use case. Or is this a matter of being purely cosmetic so only .sh files show, and get launched (so you manually load stuff via scripts)?

joolswills, 2016-02-09:
please test replacing the script with

```
#!/bin/bash
params=("$@")
pushd "/opt/retropie/emulators/rpix86"
if [[ "${params[0]}" == *.sh ]]; then
    bash "${params[@]}"
    exit
else
    ./rpix86 -a0 -f2 "${params[@]}"
fi
popd
```

joolswills, 2016-02-09:
(updated script)

BenWlson, 2016-02-09:
It's both for cosmetic purposes to only show .sh files in ES and to allow to change some of the rpix86 executable launch parameters per game.

I'll test your script above. From what I can tell, if it sees a script file, it just runs the provided script, and if it sees something else (a DOS executable) it applies that as a parameter to the rpix86 executable.

So my per game script would have to look maybe something like this:

```
#!/bin/bash
/opt/retropie/emulators/rpix86/rpix86 -a0 -f2 -noems -d/home/pi/RetroPie/roms/pc /home/pi/RetroPie/roms/pc/another/another.exe
```

joolswills, 2016-02-09:
yep

BenWlson, 2016-02-09:
Ok.. The script worked when loading my custom script, but it killed the old functionality of loading a .exe file.

How do I debug to see what the error was? It just dumps be back into ES.

BenWlson, 2016-02-09:
Wait, nevermind.. it was my typo again.. it looks like this will work perfectly.

joolswills, 2016-02-09:
ok. I have committed the change. thanks.

closed 2016-02-09 by joolswills

BenWlson, 2016-02-09:
Thanks. I will try and do my best to add to the wiki how I am generating the scripts for individual games.

Do you have any suggested online reading I can do as an introductory lesson to the scripting language used here or any good GitHub tutorials? I'm excited about contributing, but the last programming I did was 20 years ago in a high school elective.

joolswills, 2016-02-09:
I don't but most of RetroPie is bash scripting if you want to google it.

joolswills, 2016-02-09:
something like http://www.tldp.org/LDP/abs/html/ perhaps

BenWlson, 2016-02-09:
Thanks again.

Now that you've merged this into retropie, how do install it automatically? I don't see the option to install/update the rpix86 emulator in the retropie_setup menu.

joolswills, 2016-02-10:
from terminal

```
cd ~/RetroPie-Setup
sudo ./retropie_packages.sh rpix86 configure
```