RetroPie/RetroPie-Setup #1258: Gamecon script fails

TimGeyssens, 2016-02-11:
Can't get the gamecon script to finish successfully, tried both on RetroPie 3.4 and 3.5, always get failed to install...

joolswills, 2016-02-11:
Please post a log via pastebin.com (logs are stored in RetroPie-Setup/logs)

TimGeyssens, 2016-02-11:
I'm pretty new to this, any chance you can point me to a way of viewing the log files (tried tail but the log format seems compressed), thank you

TimGeyssens, 2016-02-11:
Ok found it, will try to get the whole file up here but in the meantime the error I get seems to be

ERROR: Module gamecon_gpio_rpi not found

joolswills, 2016-02-11:
can decompress it with 7zip on windows.

there is a known issue with installing the gamecon module once it has been built - see https://github.com/RetroPie/RetroPie-Setup/issues/1234 for a solution.

joolswills, 2016-02-11:
I think this is the same issue so please use the other ticket if that fix works.

TimGeyssens, 2016-02-11:
Not sure that issue is that related though, since when I run the command 

modprobe gamecon_gpio_rpi map=0,0,2,0,0,0

I also get the error message

modprobe: FATAL: Module gamecon_gpio_rpi not found

joolswills, 2016-02-11:
ok so it didn't even compile perhaps.

I will need the log to see why (I will try myself when I have some free time).

TimGeyssens, 2016-02-11:
Ok trying to get to log up here, thank you for your help!

TimGeyssens, 2016-02-11:
Yeah you are right it doesn't compile, when I check the log file for

/var/lib/dkms/db9_gpio_rpi/1.0/build/make.log

It states

ERROR: Kernel configuration is invalid
include/generated/autoconf.h or include/config/auto.conf are missing
Run 'make oldconfig && make prepare' on kernel src to fix it

joolswills, 2016-02-11:
I need to see the retropie-setup build log also please.

TimGeyssens, 2016-02-11:
Ok just trying to figure out how to get it on my machine (from the pi)

joolswills, 2016-02-11:
via an sftp client ?

TimGeyssens, 2016-02-11:
Ok got it, you can find the log attached, again thanks for the help on this  :) 
[log.txt](https://github.com/RetroPie/RetroPie-Setup/files/126248/log.txt)

joolswills, 2016-02-11:
please update retropie-setup and try again

TimGeyssens, 2016-02-11:
WIll do, is that following this guide? https://github.com/RetroPie/RetroPie-Setup/wiki/Updating-RetroPie Or which command should I execute?

joolswills, 2016-02-11:
just the updating the script bit from that guide - you don't need to do a full binary install.

TimGeyssens, 2016-02-11:
Okifo

joolswills, 2016-02-11:
if it compiles there is still the problem mentioned in the other issue - so follow that to enable the module on boot etc. I have yet to fix up the module for that (but will do)

TimGeyssens, 2016-02-11:
Hey, updated the script and now I get 

Could not install package(s) : gcc-4.7 .

joolswills, 2016-02-11:
I need a full log again. gcc-4.7 is available in the jessie repositories so it should install (it is required by the kernel headers for the gamecondriver)

joolswills, 2016-02-11:
you started from a retropie 3.4 or 3.5 image or ?

TimGeyssens, 2016-02-11:
On a 3.5 image

joolswills, 2016-02-11:
i'll check on my rpi (was testing on my dev environment)

joolswills, 2016-02-11:
works fine on my rpi with retropie 3.5 - perhaps you had a networking problem - I can't tell without seeing the log.

TimGeyssens, 2016-02-11:
Ok just a sec will get the log

TimGeyssens, 2016-02-11:
Tried to install again and now it seems to be in progress a bit longer, fingers crossed (will report back here)

TimGeyssens, 2016-02-11:
Seemed to have worked now, thanks so much for the help!

joolswills, 2016-02-11:
Welcome. I just fixed issue #1234 so please update retropie-setup and re-run the setup for gamecondriver and it should set the configs up correctly.

TimGeyssens, 2016-02-11:
Nice you rock :) :guitar: !

joolswills, 2016-02-11:
:)

closed 2016-02-11 by joolswills