RetroPie/RetroPie-Setup #1259: Neo Geo Pocket White Screen

Rocklima2, 2016-02-11:
Neo Geo Pocket load game show white screen but not error on log.

joolswills, 2016-02-11:
Works here - please use the forum for support running games - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

joolswills, 2016-02-11:
please include model of rpi and any other relevant details of your setup. thanks.

Rocklima2, 2016-02-21:
rpi model 2 , full set GoodNGPx 3.14 download emuparadise.
sorry I have update my retropie for 3.4 to 3.5 error message show on log file, but on start emulator loop infinite.
Last Blade, The - Beyond the Destiny (UE) [!].ngc
show log

name : fba-aspe
system: greyscale
catalog: 24 (sub 0)
starting pc : 0xAD8504
SWI: 00
SWI: 07
SWI: 07

infinite

InsecureSpike, 2016-02-21:
I would edit your post, and remove where you found your rom-set!

Rocklima2, 2016-02-21:
I wrote rom-set for test same roms because emulator have bug or rom-set , but I finish test on 3.5 retropie roms workings on my test.but is difficult same error with different rom-set 

Dark Arms - Beast Buster 1999 (W) [!].ngc
Evolution - Eternal Dungeons (U).ngc
Gals Fighters (UE) [!].ngc
King of Fighters R-2 (W) [!].ngc

HerbFargus, 2016-02-21:
Some romsets pending headers and other things just don't work on some emulators. both of the same roms worked fine for me so pebkac on this one. for further support use the forums as requested above. and even if it were an issue with the neo geo pocket emulator, we do not maintatain the emulator so that would be an upstream issue anyways.

closed 2016-02-21 by HerbFargus