RetroPie/RetroPie-Setup #1264: Box Art not showing when not in specific directory

biscuits99, 2016-02-14:
I'm not sure if the file locations that ES looks for the gamelist.xml are still valid here:
/opt/retropie/supplementary/emulationstation/GAMELISTS.md

However, when images are scraped using sselphs scraper via the RetroPie experimental menu it places images in:
/opt/retropie/configs/all/emulationstation/downloaded_images/{system}

These then appear fine when box art is chosen to be shown. But if your images are in:
/home/pi/RetroPie/roms/megadrive/images (as is the case if you run sselphs scraper manually without specifying a folder etc..) the boxart isnt shown (although the gamelist data including thumbnails, shows fine in ES).

This wouldnt really be an issue, but if you dont want the low-res thumbnail imagery I dont think the default method will work to show boxart.

In short, I guess I'm asking can the experimental sselph implementation accept flags so higher quality imagery can be used? Or can the boxart flag know to read the image location from the gamelist.xml?

joolswills, 2016-02-14:
it could just check in another location - but if you are saying running sselph's scraper manually gets high quality images then it goes to say it can do that :) Unless i've misunderstood?

HerbFargus, 2016-02-14:
I wonder if it would be worth adding a bit more customisability for sselphs scraper in terms of where scraped images are placed/where images are pulled from. For example if someone has their own images to scrape or if people want their images with their roms for a more modular backup etc.

biscuits99, 2016-02-14:
Ok, so workaround it to simply use the flags liek this

scraper -image_dir /opt/retropie/configs/all/emulationstation/downloaded_images/megadrive/ -image_path /opt/retropie/configs/all/emulationstation/downloaded_images/megadrive/

But yes, by default sselphs scraper gets high def which look good as box art, so I guess the experimental implementation has the thumb_only flag set - which isnt a bad idea.

joolswills, 2016-02-14:
the scraper module needs more work (configuration) yeah - https://github.com/RetroPie/RetroPie-Setup/issues/945 is still open for that reason.

joolswills, 2016-02-14:
I don't want to parse xml files for this - I will make some improvements to the scraper when I have time, but I don't want to spend lots of time on the boxart display feature, which was only added as it was something quick/simple to do (mostly - there were a few issues).

biscuits99, 2016-02-16:
For reference, another user edited the path in runcommand as a workaround:
http://blog.petrockblock.com/forums/topic/runcommand-box-art-function-not-working/#post-117127

joolswills, 2016-02-16:
the issue there is another one - in that they updated retropie-setup but didn't upgrade to the latest binaries (ES directory gets moved / symlinked)

biscuits99, 2016-02-16:
Ah I see. Thanks for implementing the feature, it makes a great aesthetic difference to the RetroPie experience. Happy to close this unless it should be open for future tweaks to the boxart config.

joolswills, 2016-02-16:
can leave it open as I will check both locations for images (just not using gameslist.xml)

joolswills, 2016-02-17:
it now checks the roms/systen/images dir first.

closed 2016-02-17 by joolswills