RetroPie/RetroPie-Setup #1266: lr-beetle-vb writes save files to BIOS directory

zerojay, 2016-02-14:
When playing Virtual Boy games using the lr-beetle-vb Libretro core, save game files (*.sav) are not written to the proper save game directory as defined in /opt/retropie/configs/all/retroarch.cfg and are instead written to /home/pi/RetroPie/BIOS/.

Relevant option from configs/all/retroarch.cfg (ROMs and save files are read and written to a CIFS share from a Linux box mounted to /home/pi/RetroPie/roms/, working on other cores perfectly fine):

```
# Save all save files (*.srm) to this directory. This includes related files like .bsv, .rtc, .psrm, etc ...
# This will be overridden by explicit command line options.
savefile_directory = /home/pi/RetroPie/roms/saves/

# Save all save states (*.state) to this directory.
# This will be overridden by explicit command line options.
savestate_directory = /home/pi/RetroPie/roms/states/
```

Full contents of /opt/retropie/configs/virtualboy/retroarch.cfg (untouched from fresh 3.5 install):

```
# Settings made here will only override settings in the global retroarch.cfg if placed above the #include line

input_remapping_directory = /opt/retropie/configs/virtualboy/

#include "/opt/retropie/configs/all/retroarch.cfg"
```

HerbFargus, 2016-02-15:
I would venture to guess its along the same lines as this issue: https://github.com/RetroPie/RetroPie-Setup/issues/1162  It would seem a lot of the mednafen (beetle) emulator cores have file path issues (which may possibly be upstream issues unrelated to retropie)

HerbFargus, 2016-02-17:
I can confirm the same issue:

Until it's addressed upstream by RetroArch I have no problem setting the default system config  in `/opt/retropie/configs/virtualboy/retroarch.cfg`

```
# Settings made here will only override settings in the global retroarch.cfg if placed above the #include line

input_remapping_directory = /opt/retropie/configs/virtualboy/

system_directory = /home/pi/RetroPie/roms/virtualboy

#include "/opt/retropie/configs/all/retroarch.cfg"
```

I made a quick and dirty setting to the configure function the script to automate placing configs above the #include line, but in order to avoid code duplication we may first have to look at integrating [inifunction](https://github.com/RetroPie/RetroPie-Setup/blob/950eaa3645112393b5adde9707303f6f5d5d0e57/scriptmodules/inifuncs.sh) for system specific retroarch configs on its own (as there are other systems that need system specific configs like  lr psx rearmed, and all the mednafen emulators at least until the issues are patched upstream) 

```
function configure_lr-beetle-vb() {
    mkRomDir "virtualboy"
    ensureSystemretroconfig "virtualboy"

    # system-specific, VB
    iniConfig " = " "" "$configdir/virtualboy/retroarch.cfg"
    iniSet "system_directory" "$romdir/virtualboy"
    local include=$(grep "^#include" "$configdir/virtualboy/retroarch.cfg")
    sed -i "/^#include/d" "$configdir/virtualboy/retroarch.cfg"
            if [[ -n "$include" ]]; then
                echo "" >>"$configdir/virtualboy/retroarch.cfg"
                echo "$include" >>"$configdir/virtualboy/retroarch.cfg"
            fi

    addSystem 1 "$md_id" "virtualboy" "$md_inst/mednafen_vb_libretro.so"
}
```

joolswills, 2016-02-18:
I would prefer to look into the upstream issue before adding setting to repoint file locations etc.

HerbFargus, 2016-07-25:
Sorted upstream

closed 2016-07-25 by HerbFargus