RetroPie/RetroPie-Setup #1271: Directlaunch Runcommand Crash

HerbFargus, 2016-02-16:
Before we actively pursue setting certain systems (e.g. Kodi see https://github.com/RetroPie/RetroPie-Setup/pull/904)  as directlaunch from emulationstation we need to

- create a function for directlaunch
- address crashing when implemented with runcommand

From testing kodi works fine when launched directly with kodi-standalone but if runcommand is used it crashes.

HerbFargus, 2016-02-21:
Just for reference this is my kodi crash log:

http://pastebin.com/KHT2eTXZ

joolswills, 2016-03-05:
I just tested and it seems fine here

launching with

```
  <system>
    <name>kodi</name>
    <fullname>Kodi</fullname>
    <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 kodi-standalone</command>
    <theme>kodi</theme>
    <directlaunch>true</directlaunch>
  </system>
```

joolswills, 2016-03-05:
```
  <system>
    <name>kodi</name>
    <fullname>Kodi</fullname>
    <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 _SYS_ kodi</command>
    <theme>kodi</theme>
    <directlaunch>true</directlaunch>
  </system>
```
works too (and gives the ability to the user to control the commandline etc)

HerbFargus, 2016-03-05:
No crash on exit? May have been addressed with your last modification to fb_restore. If you're happy I'm happy.

joolswills, 2016-03-05:
must have been that. or something else. I will switch kodi to direct launch in a PR and we can test.

HerbFargus, 2016-03-05:
Brilliant :)

HerbFargus, 2016-03-07:
Seems resolved now.

HerbFargus, 2016-03-13:
Upon further testing it seems there is an issue with directlaunch in that once implemented it will crash emulationstation when switching themes through the start menu in emulationstation.

To reproduce:

- Install updated kodi from the retropie setup script
- install a different theme than the default
- switch to the new theme - Will crash.

joolswills, 2016-03-13:
This should be opened against ES then.

HerbFargus, 2016-03-13:
opened here: https://github.com/RetroPie/EmulationStation/issues/37

closed 2016-03-13 by HerbFargus