RetroPie/RetroPie-Setup #1272: xbox one controller setup

eblabs, 2016-02-17:
There was a mention of partial xbox one controller support in the 3.4 release notes. What exactly has changed, and is there anything specific I need to do in order to use an xbox one controller?

Thanks!!

zerojay, 2016-02-17:
This should be a forum post, not a post to the issue tracker.

Our version of xboxdrv has some preliminary support for Xbox One controllers. You'll have to try it out and let us know how it works for you as I don't think any of us actually have one to test with.

joolswills, 2016-02-17:
Yeh please use the forum, thanks. - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2016-02-17 by joolswills

eblabs, 2016-02-17:
Thanks for the quick reply! Ill do some tests and follow up there.

Cheers!