RetroPie/RetroPie-Setup #1275: Xbox One support for Model 1697/Firmware 2015

zerojay, 2016-02-18:
We've had various complaints about the latest firmware for Xbox One controllers causing them to go unrecognized by xboxdrv as well as a new model of controller that goes unrecognized.

I've updated my branch of xboxdrv to include the new USB device ID numbers and it should be all that's required to get these controllers working with Retropie.

Jools, can pull from my repo (http://github.com/zerojay/xboxdrv), compile and submit the updated xboxdrv binary as I mentioned earlier today?

joolswills, 2016-02-18:
Will do, thanks. also users can always rebuild from the latest source by doing `sudo ./retropie_packages.sh xboxdrv`

joolswills, 2016-02-18:
Done. thanks

closed 2016-02-18 by joolswills

trixster1979, 2016-02-20:
This fix does not seem to work for me. I've updated the setup script, built xboxdrv from source but no luck. I've also tried installing the binary but that still doesn't help.

Attaching thr controller brings up a load of dialogue in the console, and jstest seems to recognise it too, but pressing the buttons does not toggle a response from jstest.

Emulationstation will not recognise that the pad is connected either.