RetroPie/RetroPie-Setup #1277: Xbox One controllers still not working

trixster1979, 2016-02-20:
Despite the recent update to xboxdrv to try and resolve this issue, Xbox one controllers still do not seem to be working.

I've updated the setup script, built xboxdrv from source but no luck. I've also tried installing the binary but that still doesn't help.

Attaching thr controller brings up a load of dialogue in the console, and jstest seems to recognise it too, but pressing the buttons does not toggle a response from jstest.

Emulationstation will not recognise that the pad is connected either.

zerojay, 2016-02-20:
Can you please post the output of sudo lsusb for us here?

trixster1979, 2016-02-20:
Bus 001 Device 005: ID 046d:c52e Logitech, Inc.
Bus 001 Device 007: ID 045e:02d1 Microsoft Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

zerojay, 2016-02-21:
There must be something more causing them to not work. Are you sure that the xpad driver has been blacklisted?

TroyPi, 2016-02-21:
I am having the same issue.  It does not work in the games.  How do you know if the driver was blacklisted?

zerojay, 2016-02-21:
You can check to see if the xpad module is currently loaded into the kernel with the following:

lsmod | grep xpad

TroyPi, 2016-02-21:
and its not a US keyboard so I can't even find the |

zerojay, 2016-02-21:
Maybe redirect the output of lsmod to a text file and read it?

lsmod > test.txt
nano test.txt

trixster1979, 2016-02-21:
Ok, so lsmod | grep xpad gave this:

xpad               12293  0
ff_memless       4758  1 xpad

I added the blacklist as you instructed and lsmod | grep xpad now returns no result.

However, the controller still does not function, and also jstest will now not recognise that the pad a pad is attached either.

lsusb now lists:

Bus 001 Device 005: ID 046d:c52e Logitech, Inc.
Bus 001 Device 004: ID 045e:02d1 Microsoft Corp.
Bus 001 Device 003: IDe 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Notice that Microsoft is listed as device 4 now, not device 7 as before. I do not know if this is relevant or not.

zerojay, 2016-02-21:
Not relevant but thanks for pointing it out.

Run: sudo ps aux | grep xboxdrv

trixster1979, 2016-02-21:
That gives:

pi         15508 0.0 0.2  5972  1968 tty1    S+  18:39   0:00 grep --color=auto xboxdrv

cyberneticwired, 2016-02-21:
mine displays a long string of text when i do it

trixster1979, 2016-02-21:
Does your controller work? What text is displayed?

zerojay, 2016-02-21:
trixter, you do not have xboxdrv running at all if that's the only line you get back.

trixster1979, 2016-02-21:
Right, so how do I get xboxdrv working properly?

zerojay, 2016-02-21:
RetroPie-Setup script -> 3 Configuration -> Xboxdrv -> Enable Xboxdrv.

trixster1979, 2016-02-21:
Ok, didn't realise I had to do that, sorry for being a noob.

However after enabling it (it says something about adding stuff to rc.local) and rebooting I'm still getting nothing.

trixster1979, 2016-02-21:
I can get rc.local working with sudo sh -x /etc/cr.local

Now sudo ps aux | grep xboxdrv gives me the correct stuff and jstest will see the controller again, however I still cannot toggle any buttons. The controller won't work in Emulationstation but attempting to configure a controller does seem to detect that it's attached.

Interestingly if I unplug the controller and then replug it the console says:

[ERROR] XboxdrvDaemon::launch_controller_thread(): USB device disappeared before it could be opened

zerojay, 2016-02-21:
You can try clearing out your emulationstation setup controls by deleting /home/pi/.emulationstation/es_input.cfg I believe. Then when you boot up next, it will ask you to configure controls.

trixster1979, 2016-02-21:
I think a more fundamental problem is that even though I've now seemingly got xboxdrv running correctly (and installed the fixed xboxdrv that was pushed to the retropie-setup script the other day) and the raspberry pi can 'see' the controller, neither jstest nor Emulationstation still will not recognise any button presses. So removing es_input.cfg won't make a difference until this issue is resolved.

cyberneticwired, 2016-02-21:
Mine looks similar to yours zero.

zerojay, 2016-02-21:
You can try to recompile the xboxdrv fork we got the Xbox One controller stuff from over here (https://github.com/antekone/xboxdrv) and let us know if that works for you but beyond that, I don't see anything else that I'm able to do here if it's still not working.

trixster1979, 2016-02-21:
Cyberneticwired, does your controller work?

cyberneticwired, 2016-02-21:
No, I'm invirtuteDei from the other thread.

cyberneticwired, 2016-02-22:
I also have three different Xbox one controllers I have tried and when I do the lsusb it says it is logic3? Don't know if that means anything or not

cyberneticwired, 2016-02-22:
And when I load rims on the bottom of the screen it says Xbox pad not configured but the menu won't let me configure so it detects it for sure.

zerojay, 2016-02-22:
Make sure you've configured it in EmulationStation.

cyberneticwired, 2016-02-22:
It doesn't let me when I go to configuration menu from pressing start.

zerojay, 2016-02-22:
As mentioned above, delete es_input.cfg, restart emulationstation and try again.

joolswills, 2016-02-22:
deleting es_input.cfg will remove the hook to trigger input configuration - if removing it, you will need to reinstall emulationstation from retropie-setup menu 5 to restore the default config.

trixster1979, 2016-02-22:
Ok, I'll give that a try but why doesnt jstest work correctly? Pressing any button on the controller does not toggle the relevant on/off...

cyberneticwired, 2016-02-22:
Okay, I'm going to reinstall emulation station from source to see if that might solve the configuration issue

joolswills, 2016-02-22:
There is no benefit from doing that - reinstall from binary (after removing the es_systems.cfg)

cyberneticwired, 2016-02-22:
Dumb question, but how do I remove the config file?

joolswills, 2016-02-22:
rm ~/.emulationstation/es_systems.cfg

joolswills, 2016-02-22:
sorry es_input.cfg

cyberneticwired, 2016-02-22:
Okay deleted the config file and reinstalled emulation station from binary, still no luck. Only recognizes Keyboard in welcome screen.

trixster1979, 2016-02-22:
I'm convinced the updated xboxdrv is still not picking up the controller correctly. If it did then surely jstest would work correctly..? What exactly was done to xboxdrv to try and regain compatibility with the latest controller firmware?

cyberneticwired, 2016-02-22:
My biggest thing is I know it works because I had the same firmware on my controller and I got it working two weeks ago (had to wipe sdcard so I lost it) from some forum that I found but I can't find it again..

trixster1979, 2016-02-22:
I reckon our best bet is if you can find that forum.

cyberneticwired, 2016-02-22:
I am not hoping on it, i have been searching for it since i wiped my sd card.

trixster1979, 2016-02-22:
I've deleted the es_input.cfg file too and reinstalled from binary too. No good, both my Xbox one controllers still refuse to work. Back to the drawing board.

Did someone on Reddit say xpad miight have been updated with a fix for the newest controllers? Is it worth installing that to see if it works? How would one go about doing that?

TroyPi, 2016-02-23:
You can check to see if the xpad module is currently loaded into the kernel with the following:

lsmod | grep xpad

If it returns with any line at all, likely "xpad", it means the kernel module is still loaded and that will interfere with xboxdrv working. So we can blacklist the xpad module from loading by doing the following:

sudo nano /etc/modprobe.d/raspi-blacklist.conf (Doesn't matter if it already exists or not.)

Add in the following:
 blacklist xpad

Save the file and reboot. Then try running the above lsmod | grep xpad and see if it's loaded. If not, check your controllers and see if they are working properly and recognized.

RetroPie-Setup script -> 3 Configuration -> Xboxdrv -> Enable Xboxdrv
Mine worked after all this was done, YAHOO!

cyberneticwired, 2016-02-23:
I know I have and pretty sure trixster have already tried that, but I will try again it won't hurt.

TroyPi, 2016-02-23:
what I don't like is that I lost the function of the keyboard with exception to the f4 key.  maybe I need to set it up as a controller, another problem is, I cannot change the mapping to my US k/b

cyberneticwired, 2016-02-23:
I would rather have my controller working instead of my keyboard.

cyberneticwired, 2016-02-24:
Anyone else have suggestions? :/

Mhnl1979, 2016-02-28:
I followed the steps above from TroyPi, now my xbox controller is recognised with the white light on the wire,  but it isnt recognised by the Emulation Station.  What next.? Ive updated retropie from clean install, blacklisted xpad and updated xboxdrv and enabled it.  So no result...

zerojay, 2016-02-28:
"You can try to recompile the xboxdrv fork we got the Xbox One controller stuff from over here (https://github.com/antekone/xboxdrv) and let us know if that works for you but beyond that, I don't see anything else that I'm able to do here if it's still not working."

Has no one built the version of xboxdrv from that other repository like I mentioned earlier in the thread? If that version of xboxdrv does not work for you, there's nothing I can do for you either and you'll have to wait for that person to update his xboxdrv fork to fix it. Send him this bug as well so he's at least aware there are issues. I'll see if there are any other pull requests out there that fix this for xboxdrv but I'm not currently hopeful.

Mhnl1979, 2016-02-28:
How can I do that?  What command i have to type?  Sorry being a Noor,  but im very excited to learn ☺️

Mhnl1979, 2016-03-04:
Followed the steps from the following link after  fresh install retropie 3.5,  updating script,  updating xboxdrv,  blacklisting xpad. And it worked!  Hooray! 

http://blog.petrockblock.com/forums/topic/new-xbox-one-controller-model-1697-not-supported/

zerojay, 2016-03-05:
That link is about using xpad though...?

Mhnl1979, 2016-03-05:
I think i didn't blacklisted xpad.  Id check that out. 
But the linux-headers are also installed. So maybe that did the trick.

trixster1979, 2016-03-05:
I rolled back the firmware in one of my controllers to 2014 firmware (http://mjr.mn/XboxOnePCDriversx86) but it still does not work, either with xpad or xboxdrv. Xpad does the trick of 'seeing' and it'll show up in jstest, but again button presses will not toggle and emulation station will not recognise button presses even though it sees a pad is connected.

Were it not for other people having problems I'd start to think there was something badly wrong with my setup.

Probably a stupid q, but has anyone tried burning a fresh sd card with a new RetroPie image to see if that works with the xboxdrv update?

cyberneticwired, 2016-03-05:
Man, I gave up and just invested in a wired 360 controller for $30.

zerojay, 2016-03-05:
Just making sure, but you aren't trying to use xpad and xboxdrv at the same time, right?

trixster1979, 2016-03-06:
Nope, xpad is blacklisted, the new xboxdrv is loaded. I've tried re-enabling xpad to see if it works with the rolled-back pad firmware but it still doesn't work, so I've re-blacklisted it.

LuisDiazUgena, 2016-03-08:
Hi!

I've a xbox controller that doesn't work at all. I've checked lsmod | grep xpad and there is no output.

I've blacklisted xpad and enable xboxdvr from retropiesetup.

I got kernel Linux 4.1.18-v7+ running on Rpi2. What should I do next?

Mhnl1979, 2016-03-08:
@trixster1979 I did,  but it didn't work. Even after updating the setup script. My guess is updating the linux-headers did the trick.

trixster1979, 2016-03-09:
Yeah, judging by reports that's the only way forward, but I'm assuming it's not a permanent solution as they'll be overwritten when the system is updated.

mindbl0wn, 2016-03-10:
@Mhnl1979 so I am following the link you posted but i am stuck, I cant seem to install the linux header 4.1.13-v7+,  getting errors while dpkg.
care to help?

[50 more replies]

closed 2016-10-05 by joolswills