RetroPie/RetroPie-Setup #1280: Frequent pauses lr-ppsspp

cooky069, 2016-02-21:
Iv been testing out psp emulation for the first time using lr-ppsspp and overall getting good results. I have one or two problems however, games seem to run fine then I get a pause for a few seconds and a solid yellow light on the pi before emulation resumes and the light goes out . I assume it must be while the sd card is reading. My sd card is an integral class 10 and running on a pi 2. I also only get about 14 fps, not sure if this is a related issue or not

zerojay, 2016-02-22:
Likely the system thrashing during heavy swap usage. I recently switched my RetroPie install over to a hard drive and I no longet get the heavy swaps at all when using PPSSPP.

joolswills, 2016-02-22:
it's nothing we can fix (we are not the emulator authors)  - it's an io issue though.

closed 2016-02-22 by joolswills

cooky069, 2016-02-22:
Thanks guys , that's pretty much what I was thinking . Will try a faster card or swap over to a hdd. Zerojay do you just run your roms from the hdd or have you set up the entire image to boot from it ?

zerojay, 2016-02-22:
I set up everything to boot from the hard drive. The important part, as far as I can tell, is that I created a 2GB swap partition on the hard drive (as opposed to the swap file on the sd card that you normally get). I used some of the information here to help me: http://raspberrypihobbyist.blogspot.ca/2013/07/running-from-external-hard-drive.html