RetroPie/RetroPie-Setup #1282: Genesis/Mega Drive emulator not acknowledging any cheevos

coreyhussell, 2016-02-22:
I have played a handful of Megadrive games and it logs me into the retro achievements but I can't seem to earn any on Genesis. When I go online it recognizes that I just played the games but i can't seem to get any cheevos, even though I did do them in game. They never pop up in game and don't show up on my user profile on RA: Coreyhussell. I use lr-picodrive.

zerojay, 2016-02-22:
I've noticed this as well, but the bug is upstream. Nothing to do with us to my knowledge.

coreyhussell, 2016-02-22:
Thanks for the reply. I posted on the RA forum bugs section. I will update if I hear anything.

joolswills, 2016-03-26:
Should work with lr-picodrive now - rebuild from source and test - see https://github.com/libretro/picodrive/pull/22

closed 2016-03-26 by joolswills

coreyhussell, 2016-03-27:
I updated the Retropie-Setup script, rebooted, rebuilt lr-picodrive from source, rebooted again. Then tried some Genesis games and none will load now. Goes black screen then back to game select menu. Double checked the Megadrive emulator was set to Picodrive. Can anyone confirm this?

joolswills, 2016-03-27:
it's possible recent upstream changes broke it - I'll test.

coreyhussell, 2016-03-27:
just an extra note on this.  I also tested some Master System games on lr-Picodrive with the exact same result.  Something is kicking me out of the emulator before it has a chance to boot.

CloneDevil, 2016-05-03:
Problem with crash of picodrive after rebuilding from source still occurs.

/tmp/runcommand.log:
> /opt/retropie/emulators/retroarch/bin/retroarch: symbol lookup error: /opt/retropie/libretrocores/lr-picodrive/picodrive_libretro.so: undefined symbol: __aeabi_idiv

joolswills, 2016-05-03:
That is not related to cheevos support - Please open a new ticket for build/link issues

joolswills, 2016-05-03:
(and detail configuration please - ie rpi version, version of retropie-setup script)

joolswills, 2016-05-20:
I submitted a fix for the undefined symbol error to the upstream libretro repository