RetroPie/RetroPie-Setup #1288: Change video_smooth default from true to false

objectchris, 2016-02-25:
After a bit of feedback before I do the pull request (it's a 1 line change). Currently the default for the Retropie script is for video_smooth to be true, which means the image when running retroarch cores is subjected to a bilinear filter, making it slightly blurred. I guess we all know what it looks like by now :)

Personally, I think the default to be 'false'. My reasoning is:
- If it's on it makes shaders display incorrectly, yet by default shaders are also enabled (at least, enabled to be used, if not turned on)
- 'Off' seems more like a default to me. it's the raw image and doesn't make presumptions about the user's TV setup or preference.
- Such filtering settings have typically been off in other emulator systems I use (eg, retroarch standalone). I think they are turned off in the non-libretro cores in retroarch also(?)
- There's a number of threads on the forum for people asking how to turn it off.

I realise it's easy to change in Retropie-Setup, but I guess that argument goes both ways :)

joolswills, 2016-02-25:
in reply to your points:

1) shaders are not enabled by default. Many people never bother with them.

2) it is subjective - I prefer it on (depending on which monitor I use). Others prefer it off - if you can provide a decent poll that most want it off, I'll switch the default.

3) retroarch has it on by default - check their config.

4) there were a number of threads before when it was on by default with people saying it looks blocky.

joolswills, 2016-02-25:
https://github.com/libretro/RetroArch/blob/8f73779acb0595b12008a0c4f220ead5e00a479c/config.def.h#L466

HerbFargus, 2016-02-25:
I was actually thinking about this the other day, and I think I agree with you on this point (particularly in that users have been asking about it quite a bit recently)

It really is a matter of personal preference. Not that it matters all that much but some users have been using recalbox because they think it looks better but it's really just a matter of this setting.  

But @objectchris if you want to take a screenshot with and a screenshot without and then put up a poll on the forum or something so we can get feedback from the community and we'll go from there.

joolswills, 2016-02-25:
it's worth noting that it can also depend on the monitor resolution it seems how it ends up looking or at least it looks quite different on a couple of my screens (it looks much better on, on the lower resolution monitor - the effect is less).

I'm not against switching, but I'm not going to be switching this back and forth - hence a poll is the best way which we can them point to when users complain about the default.

joolswills, 2016-02-25:
https://github.com/RetroPie/RetroPie-Setup/issues/933

objectchris, 2016-02-25:
poll created :)
http://blog.petrockblock.com/forums/topic/poll-which-default-video-mode-do-you-prefer-video_smooth-on-or-off-2/

HerbFargus, 2016-02-26:
Seems the unanimous consensus is to off so far.

Not sure if we have a time limit on the poll but it's enough to convince me.

joolswills, 2016-02-26:
Not unanimous now :-) but yes it is looking that way. I'm sure there will be complaints with whatever default.

zerojay, 2016-02-28:
Can we just create something in RetroPie-Setup to just allow easy changing of the setting so that there's little need to worry about it either way? I'm not sure what the change itself entails here.

HerbFargus, 2016-02-28:
@zerojay there already is: https://github.com/RetroPie/RetroPie-Setup/wiki/Configuration-Editor This just to switch the default that retropie ships with as the current setting makes it blurry and from the forum poll it seems most prefer it to be off rather than on.

joolswills, 2016-02-28:
could be argued that it isn't that beginner friendly, but it's easy to switch once you know how. Perhaps at some point I can have a simple / advanced mode for that.

joolswills, 2016-03-01:
@dankcushions Please be patient and wait - I am considering to do some other changes along with this in regards to the option switching - but it doesn't require you to decide when the results are in, nor is it useful to submit the 1 line change as a pull request. As my free time is involved, and I am the main developer currently, I don't like to feel under pressure as to when things happen.

dankcushions, 2016-03-01:
Again, no offence intended. I'm not looking for conflict.

joolswills, 2016-03-01:
Not offended, just wanted to clarify. Thanks.

joolswills, 2016-03-01:
regarding my changes, my current thinking is to move the manual editing to an advanced settings menu, and make the editing of common options simpler. I'll open a PR when I have something ready for a test.

joolswills, 2016-03-02:
We are going to push a new image out now, because of the RPI3 compatibility issue with 3.5 because it doesn't have the latest firmware. But this is next on my list - with hopefully some userability improvements for switching the setting.

joolswills, 2016-03-04:
for the "basic configuration" which options do you think are needed (there is an advanced menu with the rest of the options

Currently I have included

```
video_smooth
video_shader_enable
video_shader
input_overlay_enable
input_overlay
```

aspect ratio could be good (adjusting aspect ratio index) but the config editor would need some new options to support it in a user friendly way (so there is descriptive value for each key)

I can possibly add that later.

HerbFargus, 2016-03-04:
For specific control functions like dpad as analog and device index would those be worth adding? Or are controller settings better as a separate menu or function?

joolswills, 2016-03-04:
they could perhaps be added later separately (as the dpad as analog for each player will make the list overcomplicated I think).

HerbFargus, 2016-03-04:
I think you are right. Good point

joolswills, 2016-03-04:
I will do a PR now with just those few options. so I can get the smooth default changed, and then I will revisit this with some other improvements later as short on time. (such as handling the video aspect settings differently so you can choose 4:3 16:10 etc and it will configure multiple values in the configs.

dankcushions, 2016-03-04:
sounds good! i use different aspect ratios for different cores so as long as there's a 'no value'/whatever setting, I can just leave it alone.

joolswills, 2016-03-04:
https://github.com/RetroPie/RetroPie-Setup/pull/1327

joolswills, 2016-03-04:
to avoid scaling artifacts do you think we should change the runcommand render resolution to that of "config file" (which would be video output resolution) by default? Originally this was put it to help with performance, but it might not be a good default now we have rpi2 and rpi3 machines.

thoughts ?

dankcushions, 2016-03-04:
I think so! I have been using output render res on my pi and pi3 without issue. it would potentially make shaders run slower I guess? (I'm not a shader user)

joolswills, 2016-03-04:
yes, but the output resolution could be lowered then.

joolswills, 2016-03-04:
would like to get @gizmo98 thoughts on this too

gizmo98, 2016-03-05:
I think we added render res to avoid switching output resolution. Users had problems, because some displays do not support switching resolutions well or do not support modes (for example PITFT has a fixed reolution of 800x600). So if we want to do it the other way around there will be certainly users complaining about slow shaders or black screens. To avoid scaling artifacts we could also add a nativeResFactor between render res and output res.

joolswills, 2016-03-05:
ok maybe ome logic like full res on rpi2+rpi3, but half native res on rpi1 (if native res is high)

gizmo98, 2016-03-05:
I would suggest doing performance comparisons between 640x480 and 1920x1080 with demanding cores like snes9x-next + super-fx games, lr-ppsspp and lr-pcsx-reamed. Just enable show_fps to see frame times. If there is still a significant performance hit i would leave this setting as it is. Compared to most other ugly arm boards out there all pis have the advantage of resolution switching at runtime through tvservice and (render res) hardware upscaling through dispmanx.

joolswills, 2016-03-05:
you get scaling artifacts though when upscaling with dispmanx (it also adds some softness even without video_smoothing on with default config.txt) - but perhaps if rather than 640x480 we choose a resolution that is a integer fraction of original size that will help. I can have a test. 

Note I am only speaking about default settings too - the option will remain to reduce resolution. But I'm happy for more testing - anyone like to help.

gizmo98, 2016-03-05:
Softness can be reduced (or removed?) with another scaling kernel. We can add scaling kernel switching in runcommand.sh through vcgencmd. We just need to find a good scaling kernel. ;-)
https://github.com/libretro/RetroArch/issues/2641

And i think using a fraction (divider 2 or 4) of the output resolution instead of fixed 640x480 should reduce scaling artifacts.

dankcushions, 2016-03-05:
yeah the softness effect is removed if you use
````
vcgencmd scaling_kernel 0 0 0 0 0 0 0 0 1 1 1 1 255 255 255 255 255 255 255 255 1 1 1 1 0 0 0 0 0 0 0 0   1
````
which is nearest-neighbour, although you do get scaling artefacts if it's not an integer scaling, as @gizmo98 says.

There's a number of scaling options with dipsmanx - see https://github.com/raspberrypi/firmware/issues/104 (which also explains how to get the runtime vcgencmd command for each scaling mode, and how to revert back to your previous) - I didn't try the others. Maybe one of the others works better?

joolswills, 2016-03-05:
I am also thinking that the default should be that which reflects most games - rather than the few cases where there is slowdown. I will test with a default that is a fraction of the display size.

joolswills, 2016-03-05:
not sure I want to add scaling configs to runcommand also - it is getting a little over-engineered :)

joolswills, 2016-03-05:
having said that, I don't like all the extra work with calculating the resolution, and then deciding if there isn't a resolution set to use that, or a saved one. It would be much simpler to just "use output resolution" by default and for systems where there is slowdown people can a) lower video res b) lower render res

It's just simpler than trying to be too clever. I would like feedback from others on this also though.

joolswills, 2016-03-05:
config resolution by default I mean (which would be output resolution unless changed)

joolswills, 2016-03-05:
it would also simplify issues we have currently in that if you change video output resolution, it will still render at the default 640x480 currently.

joolswills, 2016-03-05:
http://blog.petrockblock.com/forums/topic/default-retroarch-render-resolution/

gizmo98, 2016-03-05:
Take into account this render resolution was set to get the same performance on all devices. Performance is predictable all over the board. Performance comes first. If performance is ok better visuals can be enabled. Threads about rpi1 resolution dependent performance issues had gone to zero after this change. So if you want to change this without proof expect users complaining about rp3.7 is worse or recalbox is much faster and such none sense. It's annoying enough to tell users every week how to set a low output resolution to get the same n64 performance as recalbox. I don't want to do this for libretro cores (again). 

If users wants better visuals they can a) increase render b) add shaders c) add overlays.

joolswills, 2016-03-05:
makes no sense to me - with my change it would require one setting to lower quality across all emulators. Currently you have to configure runcommand to stop it overriding the config to improve visuals. This is the only case where runcommand ignores config files and makes it's own default - normally it is for overrriding defaults.

We can set default render resolution in the retroarch.cfg for rpi users if you are worried - which can be easily changed. The way we have things currently makes MORE work in the majority of cases.

But let's see what the users say.

joolswills, 2016-03-05:
There will always be a few vocal users who complain - a few complaints should not override the majority of users wishes. I certainly don't care about comparisons with recalbox. Recalbox is excellent, and does it's job very well, but it is also very different from RetroPie.

joolswills, 2016-03-05:
I am not changing it without proof btw - I am changing it based on what users want - which is what we did with video smoothing.

joolswills, 2016-03-05:
perhaps a compromise would be to lower render resolution in retroarch.cfg just on rpi1 as mentioned. You will then have performance for rpi1 users, but it can be switched easily. RPI2 and above will get full quality, but can always lower it globally if they want.

gizmo98, 2016-03-05:
Yeah. Just keep rpi1 users out. Most current TVs are Full HD. rpi1 users will be doomed if you allow resolutions above 720p. 720p was the limit for most if not all libretro cores. 640x480 was the lowest common denominator.

joolswills, 2016-03-05:
I don't think you can generalise on screen size really - many tvs are HD, but many users run retropie on small devices too - plus you have things like Picades which are also not HD. I also think the userbase has changed a lot since these changes were done and many users have updated to a RPI2. 

We can put some logic in retroarch.cfg to perhaps only set a default res on RPI1 if it detects an HD screen (it could do 1/2 res), otherwise we should perhaps leave it alone. The configuration improvements I am working on will make it easy to switch output resolution from the config editor.

gizmo98, 2016-03-05:
We cannot forecast which resolution users run. If only 40% of the rpi0/1 user base has Full HD TVs, 40% of these users will be doomed. Even if the most users switched to rpi2 there are still rpi1 and rpi0 users which i would not skip out. Restrict resolutions above 720p.

Last but no least. I thought you did not want to add a sophisticated special solution for two rpi variants. ;-)

joolswills, 2016-03-05:
they won't be doomed - that's over dramatic. They will need to just lower the res in a few cases if we don't mess with defaults. I think you are clinging onto a setting that was perhaps useful back then, but implemented incorrectly (the default should never have been in runcommand).

I do not want to add a sophisticated solution - but adding a res check to retroarch.sh is a lot simpler than adding it to runcommand, which needs additional logic depending on current mode, or if a new mode is to be chosen - and I don't want to complicate runcommand further.

I actually prefer to not set it, and let the users change resolution if they need from the config editor.

joolswills, 2016-03-16:
https://github.com/RetroPie/RetroPie-Setup/pull/1384

joolswills, 2016-03-16:
BTW in regards to numbers of rpi1 users vs rpi2+ - of the 3.6 image we have ~8000 rpi1 downloads, and ~220000 rpi2+

joolswills, 2016-03-17:
ok. the PR regarding default res is merged. thanks.

closed 2016-03-17 by joolswills