RetroPie/RetroPie-Setup #1289: TFT wont display through HDMI

zonga50, 2016-02-26:
http://www.amazon.com/gp/product/B00WQMB1H6?psc=1&redirect=true&ref_=oh_aui_detailpage_o00_s00 this screen is what i have, i can't for the life of me figure out how to get retro pi to just display. the screen when powered on is just plain white.

zerojay, 2016-02-26:
It says on the page there that it needs a custom driver. Have you installed it?

HerbFargus, 2016-02-26:
This isn't really a bug or even related to retropie's code. Please use the forum for support.

https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2016-02-26 by HerbFargus

zonga50, 2016-02-26:
How do i install the drivers but still run retro pi as the OS

joolswills, 2016-02-26:
This is not a support area - please use the forum as requested. thanks.