RetroPie/RetroPie-Setup #1290: Add Xarcade2Joystick driver for X-Arcade Tankstick

Hedda, 2016-02-26:
Please consider adding Xarcade2Joystick (Xarcade to Joystick) driver or service to the RetroPie (RetroPie OS disto):

https://github.com/petrockblog/Xarcade2Jstick

X-Arcade Tankstick (a popular Arcade Game Controller from Xgaming) are awesome but by design it conflicts with a keyboard (and mouse), and Xarcade2Joystick solves this by wrapping it and presenting it as a standard joystick instead.

http://shop.xgaming.com/products/x-arcade-tankstick-trackball-usb-included

This driver (or service) linked above should make it easier to use X-Arcade Tankstick as input for use in the RetroPie distro.

Xarcade2Jstick exclusively captures the keyboard events of the X-Arcade Tankstick and maps these events to corresponding events on two virtual gamepad devices.

Xarcade2Jstick was originally written as a supplementary tool for the RetroPie Project. Using this tool allows the usage of the auto-config capability of RetroArch, which as you know is a central component of a RetroPie installation.

Hedda, 2016-02-26:
As a refernce see this project how relativle complicated it is otherwise to setup X-Arcade controllers with RetroPi

https://github.com/scottnath/retropie-with-dual-x-arcades

joolswills, 2016-02-26:
it is already included.

joolswills, 2016-02-26:
BTW the method described on that page is for Two Dual X-Arcade Joysticks - setting up one is much simpler. You can't use xarcade2jstick for 4 players with two dual xarcade sticks.

closed 2016-02-26 by joolswills