RetroPie/RetroPie-Setup #1293: Impossible pairing DualShock 3 after unpairing (it worked on retropie 3.3)

jcjdominguez, 2016-02-27:
Hi everybody. Sorry for my bad english.

I have a DualShock 3 clone, which I wasn’t able to pair after it got unpaired on RetroPie 3.1. So I had to reboot my raspberry in order to pair it again.

I upgraded to 3.3 version and it worked perfect. At least, I could stop playing, come back after 5 minutes and pair it again.

However, yesterday I upgraded to RetroPie 3.5 and I have that problem again.

Is anybody else facing that problem? Is it a bug? Where should I report it?

Thank you very much, and, one more time, excuse my bad english.

joolswills, 2016-02-29:
you might do better to ask on the forum if anyone else is having the same problem - also try starting from a fresh image to see if that makes a difference (backup your settings etc)

gizmo98, 2016-05-08:
I think this is a raspberry pi 3 and raspbian issue we solved some weeks ago (hciconfig up at startup).

closed 2016-05-08 by gizmo98