RetroPie/RetroPie-Setup #1295: Add Star Control 2/The Ur-Quan Masters

RoyalJL, 2016-02-27:
Would be nice to add The Ur-Quan Masters (the open source version of Star Control 2) to retropie.
It is a great game and can be played with a gamepad.

There is no official binaries but you can get the source code in the debian repository:
apt-get source uqm

Also you have all the extra stuff in:
apt-get install uqm-content uqm-music uqm-voice

Official website:
http://sc2.sourceforge.net/

HerbFargus, 2016-02-27:
Pretty cool game. here's a working module: Likely we'd want to compile it from source though maybe rather than adding a new repository source. idk @joolswills  what do you think?

```shell
#!/usr/bin/env bash

# This file is part of The RetroPie Project
# 
# The RetroPie Project is the legal property of its developers, whose names are
# too numerous to list here. Please refer to the COPYRIGHT.md file distributed with this source.
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
#

rp_module_id="uqm"
rp_module_desc="Ur-quan Masters (Port of DOS Game Star Control 2)"
rp_module_menus="4+"
rp_module_flags="nobin !mali !x86"

function depends_uqm() {
 # add repository to install uqm
    local config="/etc/apt/sources.list.d/jessie.list"
    if [[ ! -f "$config" ]]; then
        echo "deb http://ftp.us.debian.org/debian/ jessie contrib" >>$config
        echo "deb http://ftp.us.debian.org/debian/ jessie contrib non-free" >>$config
    fi
}

function install_uqm() {
    aptInstall --force-yes uqm uqm-content uqm-music
}

function configure_uqm() {
    mkRomDir "ports"

    addPort "$md_id" "uqm" "uqm" "uqm"
}
```

RoyalJL, 2016-02-27:
I tried the module you posted and It works perfectly, but there's a couple of things that I'd change:

The voices are missing, but that's easily fixed by changing:
```
function install_uqm() {
    aptInstall --force-yes uqm uqm-content uqm-music
}
```
with
```
function install_uqm() {
    aptInstall --force-yes uqm uqm-content uqm-music uqm-voice
}
```
Also the mouse cursor is of no use in this game and stays visible in the corner, perphaps there is a way to remove it.

HerbFargus, 2016-02-27:
yeah just add -nocursor here    `addPort "$md_id" "uqm" "Ur-quan Masters" "uqm -nocursor"`

I had forgotten about the voice package

HerbFargus, 2016-02-27:
actually no that doesnt work.  I was getting my info from here: https://www.raspberrypi.org/forums/viewtopic.php?f=2&t=5310 not sure how to remove it as of right now...

RoyalJL, 2016-02-27:
I found the solution. With -f (fullscreen) removes it.

So, the current version looks like this:

```
#!/usr/bin/env bash

# This file is part of The RetroPie Project
# 
# The RetroPie Project is the legal property of its developers, whose names are
# too numerous to list here. Please refer to the COPYRIGHT.md file distributed with this source.
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
#

rp_module_id="uqm"
rp_module_desc="Ur-quan Masters (Port of DOS Game Star Control 2)"
rp_module_menus="4+"
rp_module_flags="nobin !mali !x86"

function depends_uqm() {
 # add repository to install uqm
    local config="/etc/apt/sources.list.d/jessie.list"
    if [[ ! -f "$config" ]]; then
        echo "deb http://ftp.us.debian.org/debian/ jessie contrib" >>$config
        echo "deb http://ftp.us.debian.org/debian/ jessie contrib non-free" >>$config
    fi
}

function install_uqm() {
    aptInstall --force-yes uqm uqm-content uqm-music uqm-voice
}

function configure_uqm() {
    mkRomDir "ports"

    addPort "$md_id" "uqm" "Ur-quan Masters" "uqm -f"
}
```
I think this is it.
Thoughts?

HerbFargus, 2016-02-27:
Ah great job! looks good to me.

HerbFargus, 2016-02-28:
A cleaner way may be to do this:

```
rp_module_id="uqm"
rp_module_desc="Ur-quan Masters (Port of DOS Game Star Control 2)"
rp_module_menus="4+"
rp_module_flags="nobin !mali !x86"

function depends_uqm() {
 # add repository to install uqm
    local config="/etc/apt/sources.list.d/jessie.list"
        echo "deb http://ftp.us.debian.org/debian/ jessie contrib" >>$config
        echo "deb http://ftp.us.debian.org/debian/ jessie contrib non-free" >>$config
}

function install_uqm() {
    aptInstall --force-yes uqm uqm-content uqm-music uqm-voice
}

function configure_uqm() {
    mkRomDir "ports"

    rm /etc/apt/sources.list.d/jessie.list

    addPort "$md_id" "uqm" "Ur-quan Masters" "uqm -f"
}
```

tbh I'm not to keen on adding the jessie debian repositories as it can lead to possible issues with other things. the ideal solution would be to contact whoever maintains uqm and get them to add it to the raspbian repositories as they for some reason did not include it with raspbian. 

The other solution is for someone to patch up the builds in the source code on my cloned repo so that it fits the current standards for makefiles etc.

joolswills, 2016-03-03:
we don't want to add the repository - it will break things (btw if you call this twice you will be adding the repositories multiple times anyway).

we can build from source or something.

HerbFargus, 2016-03-03:
I've forked it here:  

https://github.com/HerbFargus/uqm

Not very standardised though. (Old code)

PR #1318 (2016-03-03): added uqm experimental module (The Ur-Quan Masters - Port of DOS game…
… Star Control 2) - thanks to Herb for support/testing. fixes #1295
commit 46047649 (2016-03-03): added uqm experimental module (The Ur-Quan Masters - Port of DOS game Star Control 2) - thanks to Herb for support/testing. fixes #1295
closed 2016-03-03 by joolswills