RetroPie/RetroPie-Setup #1296: adafruit pitft setup, cant find get_mode() in runcommand.sh

mattsmithDW, 2016-02-27:
this tutorial: https://learn.adafruit.com/running-opengl-based-games-and-emulators-on-adafruit-pitft-displays/pitft-setup

tells you how to set up a pitft screen with the retropie, however in the last step or so it tells me to add a couple lines of code to the beginning of get_mode() function in the runcommand.sh, and I have been unsuccessful at locating the get_mode() function, or a better location for the commands to go.

(it wants         
mode_new = "DMT-87"
        return
at the beginning of the function)

HerbFargus, 2016-02-27:
That tutorial is outdated as retropie is in constant development. This is not a bug. please use the forum for support for these issues.

https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

This is something that has already been addressed on the adafruit forums btw https://forums.adafruit.com/viewtopic.php?f=50&t=82865&p=421169&hilit=pitft+get_mode&sid=1b8c46db27ae88d31b319c69c57a36b9#p421169

closed 2016-02-27 by HerbFargus

joolswills, 2016-02-27:
They should not be instructing people to edit this file directly either - if their hardware needs something specific we can include it - they can submit a pull request or open an issue requesting it be added properly. I suggest emailing them this and I will be happy to help.

joolswills, 2016-02-27:
as far as I know their change is to force a mode - but we shouldn't switch mode unless requested - I need more info to include it though (from adafruit).