RetroPie/RetroPie-Setup #1297: sdl 2.0.3 -> sdl 2.0.4:  The key you just pressed is not recognized by SDL.

jimbo27, 2016-02-27:
I have upgraded sdl from 2.0.3 to 2.0.4 while installing mupen64plus through Retropie script.
After that my gamepad did not work anymore in Retroarch. My gamepad works in Emulationstation but in Retroarch my key bindings got messed up completely. I can use some buttons but they are mapped completely wrong.
I get ,any of these error messages when I start an libretrocore in Emulationstation:
INFO: The key you just pressed is not recognized by SDL. To help get this fixed, please report this to the SDL mailing list <sdl@libsdl.org> EVDEV KeyCode 545
INFO: The key you just pressed is not recognized by SDL. To help get this fixed, please report this to the SDL mailing list <sdl@libsdl.org> EVDEV KeyCode 305

I am using an Wii U Pro Controller with bluetooth in conjunction with xwiimote.
With sdl 2.0.3 everything was fine but with sdl 2.0.4 I get these errors.

I am the only one with this problem? Is this actually an RetroPie issue or an SDL issue?

joolswills, 2016-02-27:
SDL input is not used by default in retroarch on retropie (it was for one revision but we reversed it - you probably need to switch) - I would think udev driver would be better. You need to post controller configs from configs/all/retroarch-joypads and configs/all/retroarch.cfg

jimbo27, 2016-02-28:
Here is my retroarch config file: http://pastebin.com/5HEmt21Q
And my gamepad config file taken from configs/all/retroarch-joypads: http://pastebin.com/CRXrTCsN

Following your hint I checked for joystick-input-drivers. The driver in the gamepad config were alread set to udev. But in my retroarch.cfg I found "input_joypad_driver = sdl2". I changed this to "input_joypad_driver = udev". After that my gamepad works again! :+1:  :)
But I still get those error messages mentioned in my post above. I get about 10 of these just straight after selecting a romfile in Emulationstation and before I see the gray launch screen ("Press a button to configure").

joolswills, 2016-02-28:
good that it is working - I don't think I can help with the other messages as it's related to SDL2 but if everything is working I would just ignore them.

joolswills, 2016-02-28:
you could bring this up perhaps on the SDL2 forum.

joolswills, 2016-02-28:
(it doesn't recognise the inputs so it doesn't know how to map it - I don't know SDL2 enough to be able to advise)

jimbo27, 2016-02-28:
Ok, I will do so! Thank you! :)

jimbo27, 2016-02-29:
fyi: I made a bugreport at bugzilla.libsdl.org.

gizmo98, 2016-03-11:
@joolswills
How can i do a source based installation of sdl2? I think retropie setup uses a prebuild package. I just want to check if all those ps3 controller issues which pop up with rp3.6 are sdl related. Is retropie_packages.sh sdl2 the right command?

joolswills, 2016-03-11:
sudo ./retropie_packages.sh sdl2

joolswills, 2016-03-11:
I have heard that sdl 2.0.4 causes more of those messages than sdl2.0.3 - it doesn't know how to map them internally or so. ive not looked into it in detail though.

joolswills, 2016-03-11:
the mentioned SDL ticket above - https://bugzilla.libsdl.org/show_bug.cgi?id=3277

gizmo98, 2016-03-11:
Ok. I'm not happy to tell you there seems to be a regression if we use sdl 2.0.4 and ps3 controllers as well. If i install sdl 2.0.3 i can configure my usb connected controller in emulationstation again.

joolswills, 2016-03-11:
well that's annoying. Are you able to add any information to the upstream ticket ?

I will need to make some changes so we can force the downgrade of sdl2.

joolswills, 2016-03-11:
Unless it's related to some build dependencies in the newer sdl2 that got missed so it didn't include some functionality. I will do some testing.

joolswills, 2016-03-11:
a git bisect to find the breaking sdl commit would be useful also if it's related to a code change.

joolswills, 2016-03-11:
I have modified the code to be able to force the downgrade of sdl to 2.0.3 now.

gizmo98, 2016-03-11:
Have not gone any deeper. I just installed three times sdl2.0.3 and three times sdl2.0.4. If sdl2.0.4 is installed configuration does not work. If controller is already configured everything is fine. 

emulationstation.log is not informative at all. If i run controller configuration and press button X emulationstation interprets this button as `axis-16`. Button X is actually `btn 12`.

jimbo27, 2016-03-12:
At the moment I have sdl 2.0.4. How can I downgrade from sdl 2.0.4 to sdl 2.0.3? Deinstall and reinstall?

joolswills, 2016-03-12:
update the retropie-setup script and just try to reinstall retroarch from binary or do `sudo ./retropie_packages.sh sdl2 install_bin`

joolswills, 2016-03-14:
According to https://www.raspberrypi.org/blog/scummvm-sails-onto-the-raspberry-pi/#comment-1255046 - the sdl2 version of scummvm has problems on sdl 2.0.3 - can anyone confirm that - we might need to backport just those fixes to our SDL 2.0.3 whilst the SDL 2.0.4 problems are looked into.

jimbo27, 2016-03-22:
Yesterday I have installed Scummvm 1.8.0 over my prior installation of ScummVM including SDL 2.0.3. No issues with my input devices so far. Mouse and keyboard does work as designed.
My setup:
Raspberry Pi 2, kernel 4.1.19, bluetooth (bluez). Keyboard and mouse connected via bluetooth.

joolswills, 2016-04-09:
I have bisected this to https://hg.libsdl.org/SDL/rev/6c86d3216f6a

from this ticket
https://bugzilla.libsdl.org/show_bug.cgi?id=2719

I will sort out an sdl 2.0.4 with this reverted, and I will report it upstream on the ticket referenced earlier.

joolswills, 2016-04-09:
I'm not 100% sure yet it is the same problem with my ps3 controller and the original post with the wii U - but I am uploading some new sdl binaries now which can be tested

joolswills, 2016-04-09:
@jimbo27 please can you test out installing these packages

```
cd
wget http://files.retropie.org.uk/binaries/jessie/rpi2/libsdl2-2.0-0_2.0.4+2rpi_armhf.deb
wget http://files.retropie.org.uk/binaries/jessie/rpi2/libsdl2-dev_2.0.4+2rpi_armhf.deb
sudo dpkg -i libsdl2-2.0-0_2.0.4+2rpi_armhf.deb libsdl2-dev_2.0.4+2rpi_armhf.deb
```

@gizmo - feel free to test also - they def should solve the ps3 controller configuration issue, but just depends if that is the same problem @jimbo27 had

HerbFargus, 2016-04-10:
@joolswills I can confirm that my wiiu pro controller works with the above changes

joolswills, 2016-04-11:
@HerbFargus thanks for that - did you try with the older 2.0.4 to check that it is broken in the same way ? Just to be 100%

HerbFargus, 2016-04-11:
Yes I made sure to test with 2.04 first and it was indeed broken

joolswills, 2016-04-11:
cheers. I will push out the update and update the upstream ticket.

closed 2016-04-12 by joolswills

gizmo98, 2016-04-12:
My PS3 controller runs as well.

joolswills, 2016-04-12:
thanks @gizmo98

jimbo27, 2016-04-16:
Sorry for being late.

Today I updated the RetroPie script to 3.7. I reinstalled sdl2 and got sdl 2.0.4+2rpi.
I still get those error messages mentioned in my first post. I get those messages right after launching any emulator through Emulationstation.

runcommand.log: http://pastebin.com/pV045cg7

joolswills, 2016-04-16:
@HerbFargus tested with the same controller and it worked for him, so I'm not sure what the problem is - but does the controller work - despite outputting the errors ?

jimbo27, 2016-04-16:
Yes, It does work. Tested with Emulationstation, Retroarch, ScummVM, Mupen64Plus.
Only my runcommand.log gets filled with those error messages.

movisman, 2016-04-20:
Hi,
I have exactly the same issue with my Picade configured as a gamepad using SDL2 driver. Everything works fine but the screen gets littered with these errors. Fine to ignore then, but would be great to work out how to either suppress or remove them.

joolswills, 2016-04-20:
Must be some change from 2.0.3 -> 2.0.4 but without one of the controllers I cannot work out what changed in SDL2 to cause this.

joolswills, 2016-04-20:
actually, I do have a picade that I could flash the new firmware on. I wouldn't think you want to use the sdl2 driver in retroarch btw - the udev driver "should" be better. We had plenty of people reporting trouble with the sdl2 driver. I'll try this when I get a chance.

joolswills, 2016-04-20:
This can probably be fixed in the picade firmware too - as I have plenty of USB gamepads that don't cause this problem - maybe SDL isn't actually at fault but the gamepad isn't following specs in terms of input ids etc.

movisman, 2016-04-20:
Thanks, having 'udev' set in the retroarch.cfg does not work for me since I flashed my Picade with the gamepad firmware - it simply doesn't respond in games unless I use SDL. The firmware I currently use with the Picade is via this link:
http://forums.pimoroni.com/t/new-2-x-configurable-keyboard-gamepad-firmware/1881

Gadgetoid mentions a few paragraphs in about the console errors appearing on SDL2.x, but not SDL 1.x.

I have tried to suppress the emulationstation console output via the 10-emulationstation.sh file, and while I can get it to stay cleared for 'normal' use, if I access any console based menus like runcommand, these get stuck on the console screen when they are exited.

movisman, 2016-04-20:
PS. Were you on 2.0.3 up until very recently? I seem to remember it was. In which case I still got the console errors on 2.0.3.

joolswills, 2016-04-20:
ok so this may be unrelated to the sdl2.0.3 -> sdl2.0.4 changes then, but I suspect if it doesn't work with the udev driver there is something not right with the firmware but would need to test further.

movisman, 2016-04-20:
Thanks, yes it is a bit strange. I was expecting the console errors prior to install as per Gadgetoid's comment on the thread:
_"I tried sdl-jstest from here ( https://github.com/Grumbel/sdl-jstest3 ) to confirm buttons were being read- it works fine in SDL 1.2, but for some completely insane reason SDL 2.0 emits console warnings about not supporting certain keycodes whenever a Picade Gamepad button is hit."_

Putting up with the warnings is definitely better than having the Picade mapped as a keyboard for a few reasons, but it would be nice to get rid of these warnings, or at least silence them so it looks really slick. I will post on the Picade forum also to see what they think regarding the errors.

joolswills, 2016-04-20:
Another option is if it all works, despite the warnings, is to disable them in our branch - which would be a simple "workaround" for the annoyance. unfortunately despite the error saying to report it upstream, there doesn't seem to be much feedback when doing so from the  upstream devs.

joolswills, 2016-04-20:
Sorry which is basically what you were saying. misread :-) I think I will do that actually.

movisman, 2016-04-20:
Hi,
Yes, everything works perfectly using the sdl2 driver, no problems whatsoever. It's just those console warnings which appear. If you are able to disable those warnings in your branch - that would be excellent?

movisman, 2016-04-21:
I forgot to say last night, if you can disable the warnings in your branch and need me to test on my Picade, just let me know and i'll be happy to do so.

joolswills, 2016-04-21:
thanks. Will throw up a package for you asap.

joolswills, 2016-04-21:
Please try these sdl2.0.4 packages

http://malus.exotica.org.uk/~buzz/retropie/

jimbo27, 2016-04-21:
Does it make sense for me to test them, too? I'd welcome a suppression of those sdl warnings. Or is this version of sdl a special one just for Picade?

joolswills, 2016-04-21:
feel free.

movisman, 2016-04-21:
Hi,
Thanks a lot for the package. Just installed them one after the other, quick reboot, no more warnings!! Amazing job, thanks. Have you just suppressed these particular ones or silenced all SDL related console output?

joolswills, 2016-04-21:
Just those ones. I'll put it into retropie then with these changes.

jimbo27, 2016-04-21:
Confirmed. No more annoying warnings with this sdl package. Great! Thanks a lot! :+1:

movisman, 2016-04-21:
Great, it would be very useful to a few others i'm sure who get annoyed by these messages. Thanks a lot for your help with this. Is the sdl driver updated/reinstalled when the setup script is updated by the way?

joolswills, 2016-04-22:
New sdl2 packages have now been built and will be installed if retropie-setup is updated and then something that uses sdl2 is re-installed (or a full binary install is done).

movisman, 2016-04-23:
Understood, thanks a lot for your help! :+1: