RetroPie/RetroPie-Setup #1299: Github slow download speeds

thetonyleone, 2016-02-28:
Hello, 

I was just signing up to download retropie and the download speeds from GitHub are really slow. Im only getting 35KB/sec. It will take 6 hours to download the whole image. Anyone else having problems?

HerbFargus, 2016-02-28:
Likely your network. no problems here

thetonyleone, 2016-02-28:
I checked multiple downloads even from Microsoft and other sites plug speed tests and dont see a problem on mine. 
[image]

joolswills, 2016-02-28:
Maybe a problem between you and the particular machine serving up the data. Try it again - you might get a different machine in the CDN. This issue tracker is for bugs with RetroPie - we are not responsible for Githubs hosting - use the forum for further discussion.

https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2016-02-28 by joolswills