RetroPie/RetroPie-Setup #1312: USB Audio (Quick Fix)

jrcharney, 2016-03-01:
There's a lot of old documentation on how to set this up.  I just though I would bring to everyone's attention how to fix this, espeically if you are using a USB Audio dongle in leu of the Raspberry Pi's scratch 3.5 mm audio jack. (Blame the manufacturer of that part. Hopefully the foundation will fix it when the next Pi comes out.)

So instead of using one of those large shield with the RCA jacks on it, a small [$2 USB Audio Dongle](http://www.amazon.com/Channel-External-Sound-Audio-Adapter/dp/B007HISGRW) and a few commands.

1.    At startup, press <kbd>F4</kbd> to go to the console and enter these commands.
2.    Assuiming the USB Audio dongle is plugged in, type `lsusb`, and look for some device with "C-Media Electronics, Inc. Audio Adapter" in it
3.   Not a neccssary step but one that you should note. Type, `amixer`.  You'll see the default bcm2835 set up. Our device is not set up.
4.    Find our card number.  More than likely, the "C-Media USB Headphone Set" will be set to `Card 1`.  We'll need to remember this number for the next step.
5.    Use `nano` to create `/etc/asound.conf` with this content.  Press <kbd>Ctrl</kbd>+<kbd>X</kbd> followed by <kbd>Y</kbd> when finished.  Replace the card number in the code below with the correct card number if it is other than `1`.

     ```
    pcm.!default {
     type hw card 1
    }
    ctl.!default {
     type hw card 1
    }
    ```

6.    `sudo reboot`.  When Emulation Station or a video game is played, you will start to notice sound.

7.    If you want to test stereo, <kbd>F4</kbd> again to do a speaker test using `speaker-test -c2 hw:Set,0` where `-c2` indicates the number of channels. If you have just one speaker, use `-c1` instead.  You should hear a hiss in each speaker separately.  This test will go on continuously until you press <kbd>Ctrl</kbd>+<kbd>C</kbd>.

There is a pitfall with these instructions, and that is I don't know what will happen if you plug this into HDMI later.  If anyone knows, please share what happens.

zerojay, 2016-03-01:
This should be posted to the forum or wikI, not in the bug tracker.

jrcharney, 2016-03-01:
My apologies. I did not know that other people may edit the wiki. I have added this information to the "Sound Issues" page.

closed 2016-03-01 by jrcharney

HerbFargus, 2016-03-01:
A wiki by definition is a place for collaborative editing. Your contributions are appreciated as the lovely buzzing noise for composite output has plagued many a user. 

The recent interview with eben upton on the pi podcast seemed to suggest it isn't an issue with the Rpi 3 or at least isn't as big of an issue. I won't know for sure  until it arrives.

jrcharney, 2016-03-01:
@HerbFargus Thank you. However, I still do get a little bit of a buzz.  Maybe it is a bad Sound card, or the fact that the sound card is surounded by three other wireless broadcasting devices (WiFi, Bluetooth, 2.4 GHz keyboard receiver).  To be sure, I'm ordering another USB Audio dongle.  I may have to try out `disable_audio_dither` as well, which the guy just above the [section that I wrote](https://github.com/RetroPie/RetroPie-Setup/wiki/Sound-Issues#usb-audio) addressed.

I hope Eben Upton is right.  The appeal of the Pi 3 is the onboard Wifi and Bluetooth would free up a lot of USB space.  If only the upgrade from Pi 2 to Pi 3 was simple as transferring the microSD card, but the Pi 3 has a 64-bit arch.