RetroPie/RetroPie-Setup #1314: Official support for Hardkernel ODROID-C1+ and ODROID-C2?

Hedda, 2016-03-02:
Any plans by the RetroPie team to officially support Hardkernel's ODROID-C1 and/or ODROID-C2?

I would think that supporting ODROID-C1+ and ODROID-C2 would not be that much different than Raspberry Pi devices?

Maybe the RetroPie team could get a sponsoring deal with Hardkernel?

What makes these very attractive devices to RetroPie users is they offer more power at same price as Raspberry Pi.

If too late to begin supporting the Amlogic S805 based ODROID-C1+ then ODROID-C2 is Amlogic S905 based.

joolswills, 2016-03-02:
We have basic Odroid-c1 support already. some info here - https://github.com/retropie/retropie-setup/wiki/Odroid

joolswills, 2016-03-02:
I would add that the ODroid c1 Mali GPU drivers are not that great, and you can only have either X or accelerated framebuffer currently afaik, not both, and the software support isn't as good as that of the RPI from my experience.

Hedda, 2016-03-02:
Cool. Personally I'm really only interested in ODROID-C2 though.

Maybe a parnership with Hardkernel could sort stuff like drivers?

HerbFargus, 2016-03-02:
I wouldn't hold your breath on that

joolswills, 2016-03-03:
if they were to send me some boards (I have an odroid-c1) then I could look at improving compatibility, but I wouldn't want to be under any obligation. They need to contact us though.

Twitch0815, 2016-03-05:
Just adding a +1 
Have two c1+ I wish I could trade for raspi 3's but would be nice to use them as emulation station. 

Are you guys active working on the binary installation method as opposed to just source? 
Maybe some better documentation about what will compile and what will not?

HerbFargus, 2016-03-05:
There won't be any odroid binaries. you will have to install them from source and only some libretro emulators are supported, It's unlikely to ever get the same level of support as the pi because odroid doesnt have the same software support that the raspberry pi foundation provides.

closed 2016-03-26 by joolswills