RetroPie/RetroPie-Setup #1315: Add touch support for official 7" display

LuisDiazUgena, 2016-03-02:
Hi!

Thanks for all your work! It's amazing to have this greats pieces of hardware in the Pi!

I recently got the official rpi display and the image quality is pretty great. It could be a good enhancement to have the touch interface working.

Grettings.

joolswills, 2016-03-03:
None of the devs have one of these and I don't have interest in this - but if someone creates some input driver for it where it emulates a joypad or something I will be happy to include it.

Dimensional, 2016-07-27:
I whole heartedly would love to see this implimented. If the DS emulator can be optimized and run faster, then the touch screen at 90 degrees could make for the perfect dual screen setup with touch capabilities for DS games that require it, like Pokemon Dash, Zelda Phantom Hourglass and Spirit Tracks for example.

joolswills, 2016-07-27:
I don't see that ever happening. I am going to close this ticket, as it isn't something I will implement sorry.

closed 2016-07-27 by joolswills