RetroPie/RetroPie-Setup #1316: lr-mame build fails

HerbFargus, 2016-03-02:
Log here:

[rps_2016-03-02_020434.log.gz](https://github.com/RetroPie/RetroPie-Setup/files/155453/rps_2016-03-02_020434.log.gz)

took 7 hours before it failed.

joolswills, 2016-03-02:
you ran out of memory  - tbh I really only added this for x86 but you will need to add more swap or something and / or reduce gpu mem.

HerbFargus, 2016-03-02:
Fair enough. (x86 only is prob wise)

closed 2016-03-02 by HerbFargus

joolswills, 2016-03-02:
you can change the module and experiment adding "rpSwap on 1000 (or higher)" and see if that helps.

joolswills, 2016-03-02:
be prepared to wear out your flash :-)

HerbFargus, 2016-03-02:
I've got plenty, they are disposable. I'll try it out with the newer releases of commander genius as well and see if that makes a difference provided the memory was the issue with the GCC crashing.