RetroPie/RetroPie-Setup #1317: lr-imame4all compilation error on pi3: Error: swp{b} use is obsoleted for ARMv8 and later

dankcushions, 2016-03-02:
````
cc -DGP2X -DALIGN_INTS -DALIGN_SHORTS -DINLINE="static __inline"  -DMAME_UNDERCLOCK -DMAME_FASTSOUND -DENABLE_AUTOFIRE -DBIGCASE -D__LIBRETRO__ -DIS_ARM=1 -DWANT_LIBCO -I./src/libretro/libretro-common/include -ffast-math -fPIC  -Isrc/zlib -Wno-sign-compare -Wunused -Wpointer-arith -Waggregate-return -Wshadow -Werror=implicit-function-declaration -I./src -I./src/libretro -I./src/libretro/libretro-common/include  -DHAS_Z80=1 -DHAS_8080=1 -DHAS_8085A=1 -DHAS_M6502=1 -DHAS_M65C02=1 -DHAS_M6510=1 -DHAS_N2A03=1 -DHAS_H6280=1 -DHAS_I86=1 -DHAS_I186=1 -DHAS_V20=1 -DHAS_V30=1 -DHAS_V33=1 -DHAS_I8035=1 -DHAS_I8039=1 -DHAS_I8048=1 -DHAS_N7751=1 -DHAS_M6800=1 -DHAS_M6801=1 -DHAS_M6802=1 -DHAS_M6803=1 -DHAS_M6808=1 -DHAS_HD63701=1 -DHAS_NSC8105=1 -DHAS_M6805=1 -DHAS_M68705=1 -DHAS_HD63705=1 -DHAS_M6805=1 -DHAS_HD6309=1 -DHAS_M6809=1 -DHAS_KONAMI=1 -DHAS_M68000=1 -DHAS_M68010=1 -DHAS_M68EC020=1 -DHAS_M68020=1 -DHAS_T11=1 -DHAS_S2650=1 -DHAS_TMS34010=1 -DHAS_TMS9980=1 -DHAS_Z8000=1 -DHAS_TMS320C10=1 -DHAS_CCPU=1 -DHAS_ADSP2100=1 -DHAS_ADSP2105=1 -DHAS_MIPS=1 -DHAS_CYCLONE=1 -DHAS_DRZ80=1 -DHAS_AY8910=1 -DHAS_YM2203=1 -DHAS_YM2151=1 -DHAS_YM2608=1 -DHAS_YM2610=1 -DHAS_YM2610B=1 -DHAS_YM2612=1 -DHAS_YM3438=1 -DHAS_YM2413=1 -DHAS_YM3812=1 -DHAS_YM3526=1 -DHAS_Y8950=1 -DHAS_CUSTOM=1 -DHAS_YMZ280B=1 -DHAS_POKEY=1 -DHAS_QSOUND=1 -DHAS_SN76477=1 -DHAS_SN76496=1 -DHAS_TMS36XX=1 -DHAS_TMS5220=1 -DHAS_SEGAPCM=1 -DHAS_DISCRETE=1 -DHAS_DAC=1 -DHAS_ADPCM=1 -DHAS_OKIM6295=1 -DHAS_SAMPLES=1 -DHAS_ASTROCADE=1 -DHAS_NES=1 -DHAS_NAMCO=1 -DHAS_VLM5030=1 -DHAS_MSM5205=1 -DHAS_UPD7759=1 -DHAS_HC55516=1 -DHAS_K005289=1 -DHAS_K007232=1 -DHAS_K051649=1 -DHAS_K053260=1 -DHAS_K054539=1 -DHAS_RF5C68=1 -DHAS_CEM3394=1 -DHAS_C140=1 -O2 -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -pipe -O3 -c src/cpu/m68000_cyclone/cyclone.s -o src/cpu/m68000_cyclone/cyclone.o
In file included from src/libretro/libretro-common/libco/libco.c:14:0:
src/libretro/libretro-common/libco/armeabi.c:39:13: warning: ‘crash’ defined but not used [-Wunused-function]
 static void crash(void)
             ^
cc -DGP2X -DALIGN_INTS -DALIGN_SHORTS -DINLINE="static __inline"  -DMAME_UNDERCLOCK -DMAME_FASTSOUND -DENABLE_AUTOFIRE -DBIGCASE -D__LIBRETRO__ -DIS_ARM=1 -DWANT_LIBCO -I./src/libretro/libretro-common/include -ffast-math -fPIC  -Isrc/zlib -Wno-sign-compare -Wunused -Wpointer-arith -Waggregate-return -Wshadow -Werror=implicit-function-declaration -I./src -I./src/libretro -I./src/libretro/libretro-common/include  -DHAS_Z80=1 -DHAS_8080=1 -DHAS_8085A=1 -DHAS_M6502=1 -DHAS_M65C02=1 -DHAS_M6510=1 -DHAS_N2A03=1 -DHAS_H6280=1 -DHAS_I86=1 -DHAS_I186=1 -DHAS_V20=1 -DHAS_V30=1 -DHAS_V33=1 -DHAS_I8035=1 -DHAS_I8039=1 -DHAS_I8048=1 -DHAS_N7751=1 -DHAS_M6800=1 -DHAS_M6801=1 -DHAS_M6802=1 -DHAS_M6803=1 -DHAS_M6808=1 -DHAS_HD63701=1 -DHAS_NSC8105=1 -DHAS_M6805=1 -DHAS_M68705=1 -DHAS_HD63705=1 -DHAS_M6805=1 -DHAS_HD6309=1 -DHAS_M6809=1 -DHAS_KONAMI=1 -DHAS_M68000=1 -DHAS_M68010=1 -DHAS_M68EC020=1 -DHAS_M68020=1 -DHAS_T11=1 -DHAS_S2650=1 -DHAS_TMS34010=1 -DHAS_TMS9980=1 -DHAS_Z8000=1 -DHAS_TMS320C10=1 -DHAS_CCPU=1 -DHAS_ADSP2100=1 -DHAS_ADSP2105=1 -DHAS_MIPS=1 -DHAS_CYCLONE=1 -DHAS_DRZ80=1 -DHAS_AY8910=1 -DHAS_YM2203=1 -DHAS_YM2151=1 -DHAS_YM2608=1 -DHAS_YM2610=1 -DHAS_YM2610B=1 -DHAS_YM2612=1 -DHAS_YM3438=1 -DHAS_YM2413=1 -DHAS_YM3812=1 -DHAS_YM3526=1 -DHAS_Y8950=1 -DHAS_CUSTOM=1 -DHAS_YMZ280B=1 -DHAS_POKEY=1 -DHAS_QSOUND=1 -DHAS_SN76477=1 -DHAS_SN76496=1 -DHAS_TMS36XX=1 -DHAS_TMS5220=1 -DHAS_SEGAPCM=1 -DHAS_DISCRETE=1 -DHAS_DAC=1 -DHAS_ADPCM=1 -DHAS_OKIM6295=1 -DHAS_SAMPLES=1 -DHAS_ASTROCADE=1 -DHAS_NES=1 -DHAS_NAMCO=1 -DHAS_VLM5030=1 -DHAS_MSM5205=1 -DHAS_UPD7759=1 -DHAS_HC55516=1 -DHAS_K005289=1 -DHAS_K007232=1 -DHAS_K051649=1 -DHAS_K053260=1 -DHAS_K054539=1 -DHAS_RF5C68=1 -DHAS_CEM3394=1 -DHAS_C140=1 -O2 -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -pipe -O3 -c src/cpu/m68000_cyclone/c68000.c -o src/cpu/m68000_cyclone/c68000.o
cc -DGP2X -DALIGN_INTS -DALIGN_SHORTS -DINLINE="static __inline"  -DMAME_UNDERCLOCK -DMAME_FASTSOUND -DENABLE_AUTOFIRE -DBIGCASE -D__LIBRETRO__ -DIS_ARM=1 -DWANT_LIBCO -I./src/libretro/libretro-common/include -ffast-math -fPIC  -Isrc/zlib -Wno-sign-compare -Wunused -Wpointer-arith -Waggregate-return -Wshadow -Werror=implicit-function-declaration -I./src -I./src/libretro -I./src/libretro/libretro-common/include  -DHAS_Z80=1 -DHAS_8080=1 -DHAS_8085A=1 -DHAS_M6502=1 -DHAS_M65C02=1 -DHAS_M6510=1 -DHAS_N2A03=1 -DHAS_H6280=1 -DHAS_I86=1 -DHAS_I186=1 -DHAS_V20=1 -DHAS_V30=1 -DHAS_V33=1 -DHAS_I8035=1 -DHAS_I8039=1 -DHAS_I8048=1 -DHAS_N7751=1 -DHAS_M6800=1 -DHAS_M6801=1 -DHAS_M6802=1 -DHAS_M6803=1 -DHAS_M6808=1 -DHAS_HD63701=1 -DHAS_NSC8105=1 -DHAS_M6805=1 -DHAS_M68705=1 -DHAS_HD63705=1 -DHAS_M6805=1 -DHAS_HD6309=1 -DHAS_M6809=1 -DHAS_KONAMI=1 -DHAS_M68000=1 -DHAS_M68010=1 -DHAS_M68EC020=1 -DHAS_M68020=1 -DHAS_T11=1 -DHAS_S2650=1 -DHAS_TMS34010=1 -DHAS_TMS9980=1 -DHAS_Z8000=1 -DHAS_TMS320C10=1 -DHAS_CCPU=1 -DHAS_ADSP2100=1 -DHAS_ADSP2105=1 -DHAS_MIPS=1 -DHAS_CYCLONE=1 -DHAS_DRZ80=1 -DHAS_AY8910=1 -DHAS_YM2203=1 -DHAS_YM2151=1 -DHAS_YM2608=1 -DHAS_YM2610=1 -DHAS_YM2610B=1 -DHAS_YM2612=1 -DHAS_YM3438=1 -DHAS_YM2413=1 -DHAS_YM3812=1 -DHAS_YM3526=1 -DHAS_Y8950=1 -DHAS_CUSTOM=1 -DHAS_YMZ280B=1 -DHAS_POKEY=1 -DHAS_QSOUND=1 -DHAS_SN76477=1 -DHAS_SN76496=1 -DHAS_TMS36XX=1 -DHAS_TMS5220=1 -DHAS_SEGAPCM=1 -DHAS_DISCRETE=1 -DHAS_DAC=1 -DHAS_ADPCM=1 -DHAS_OKIM6295=1 -DHAS_SAMPLES=1 -DHAS_ASTROCADE=1 -DHAS_NES=1 -DHAS_NAMCO=1 -DHAS_VLM5030=1 -DHAS_MSM5205=1 -DHAS_UPD7759=1 -DHAS_HC55516=1 -DHAS_K005289=1 -DHAS_K007232=1 -DHAS_K051649=1 -DHAS_K053260=1 -DHAS_K054539=1 -DHAS_RF5C68=1 -DHAS_CEM3394=1 -DHAS_C140=1 -O2 -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -pipe -O3 -c src/cpu/z80_drz80/drz80.s -o src/cpu/z80_drz80/drz80.o
src/cpu/z80_drz80/drz80.s: Assembler messages:
src/cpu/z80_drz80/drz80.s:4085: Error: swp{b} use is obsoleted for ARMv8 and later
src/cpu/z80_drz80/drz80.s:4087: Error: swp{b} use is obsoleted for ARMv8 and later
src/cpu/z80_drz80/drz80.s:5141: Error: swp{b} use is obsoleted for ARMv8 and later
src/cpu/z80_drz80/drz80.s:5143: Error: swp{b} use is obsoleted for ARMv8 and later
src/cpu/z80_drz80/drz80.s:5145: Error: swp{b} use is obsoleted for ARMv8 and later
Makefile:219: recipe for target 'src/cpu/z80_drz80/drz80.o' failed
make: *** [src/cpu/z80_drz80/drz80.o] Error 1
make: *** Waiting for unfinished jobs....
/home/pi/RetroPie-Setup/scriptmodules/emulators
Could not successfully build Arcade emu - iMAME4all (based on MAME 0.37b5) port for libretro (/home/pi/RetroPie-Setup/tmp/build/lr-imame4all/mame2000_libretro.so not found).
````
see http://ctrlv.it/id/4749/4104641947 for the full log

Looks like it's this issue: http://stackoverflow.com/questions/13584091/arm-warning-swpb-use-is-deprecated-for-this-architecture - don't entirely follow the answers, though :)

zerojay, 2016-03-02:
This is going to be an upstream issue then... and not likely to get fixed at all unless someone decides to take a shot at assembler.

dankcushions, 2016-03-02:
if it's to do with gcc versions it's not upstream I think?

joolswills, 2016-03-03:
It's an upstream issue. You can either use the binary or adjust the compiler flags with sudo CFLAGS="" ./retropie_packages.sh etc.

joolswills, 2016-03-03:
(note we have a fork of this, but it is not managed by us - we just removed some makefile stuff for retropie - the rpi port is hosted at https://sourceforge.net/projects/mame4allpi/ - but there is not much/any development happening afaik)

joolswills, 2016-03-03:
the easiest fix for now is for us to override the RPI3 CFLAGS for this module. which I will do in the meantime (I doubt anyone is going to fix drz80)

joolswills, 2016-03-03:
sorry i was talking about mame4all not the libretro core.

joolswills, 2016-03-03:
(You could also disable the arm assembler optimisations or the arm z80 core  in particular - if swp is being emulated via a trap as the stackoverflow says, it could cause overhead - would need to be tested if performance is better or worse)

joolswills, 2016-03-03:
workaround committed

closed 2016-03-03 by joolswills

dankcushions, 2016-03-03:
Totally works! cheers jools :) I raised upstream - maybe there's some bored asm programmers who want to work on ancient mame cores :) https://github.com/libretro/mame2003-libretro/issues/18

fruit-bat, 2017-01-14:
diff --git a/src/cpu/z80_drz80/drz80.s b/src/cpu/z80_drz80/drz80.s
index 9d41f0d..3c687fe 100644
--- a/src/cpu/z80_drz80/drz80.s
+++ b/src/cpu/z80_drz80/drz80.s
@@ -4082,9 +4082,13 @@ opcode_0_7:
 ;@EX AF,AF'
 opcode_0_8:
        add r1,cpucontext,#z80a2
-       swp z80a,z80a,[r1]
+       ldr r0,[r1]^M
+       str z80a,[r1]^M
+       mov z80a,r0^M
        add r1,cpucontext,#z80f2
-       swp z80f,z80f,[r1]
+       ldr r0,[r1]^M
+       str z80f,[r1]^M
+       mov z80f,r0^M
        fetch 4
 ;@ADD HL,BC
 opcode_0_9:
@@ -5138,11 +5142,17 @@ opcode_D_8:
 ;@EXX
 opcode_D_9:
        add r1,cpucontext,#z80bc2
-       swp z80bc,z80bc,[r1]
+       ldr r0,[r1]^M
+       str z80bc,[r1]^M
+       mov z80bc,r0    ^M
        add r1,cpucontext,#z80de2
-       swp z80de,z80de,[r1]
+       ldr r0,[r1]^M
+       str z80de,[r1]^M
+       mov z80de,r0     ^M
        add r1,cpucontext,#z80hl2
-       swp z80hl,z80hl,[r1]
+       ldr r0,[r1]^M
+       str z80hl,[r1]^M
+       mov z80hl,r0      ^M
        fetch 4
 ;@JP C,$+3
 opcode_D_A:

commit 0fcb5554 (2016-03-03): workaround building mame4all/lr-imame4all from source on armv8 due to obsoleted asm in drz80 - use armv7 target instead - fixes #1317