RetroPie/RetroPie-Setup #1333: ps3 pairing issue / solution

joolswills, 2016-03-05:
Got this from a user on the forum 

> first of all all! 
> you are doing a wonderful job!
> 
> as retropie went over to jessie, i had alot of
> issues of pairing ps3 controllers after a reboot.
> i had to connect the usb cable to get the
> controller to pair.
> but now i found a little quickfix, not a solution.
> 
> since the sixad is started but refuses to pair
> controller after reboot.
> and sudo sixad –start (even as it already
> started it will pair the controller directly)
> 
> i just added sixad –start in rc.local so it will
> be started twice.
> 
> now controller pairs on every reboot/cold start
> 

I need to test myself, but putting this here in case there is an issue - until I have had a change to check. I've read a few things recently with ps3 controllers, so not sure if we have a bug or not.

BenWlson, 2016-03-05:
I have an authentic PS3 controller that I pair with a Kinivo USB Bluetooth dongle and I have had zero issues. Connects every time after initial pairing. No need to reconnect with USB. Only complaint is that it does not timeout due to inactivity. The PS3 controller will stay connected until the battery dies if left alone.

joolswills, 2016-03-05:
thanks. I will test with my controller as it also worked last time I checked. Not sure why some users are having trouble - problems with specific dongles perhaps.

gizmo98, 2016-03-05:
I had a cold boot issue since we use raspbian jessie. It seems like the hci device is not up issue is back. Connecting a controller over usb (which triggers hciconfig hci0 up) helps to pair controller again. sixad --start runs hciconfig hci0 up as well. I thought the problem was gone with the latest changes. But my new rpi3 still seems to have a pairing issue if i try to pair "before" emulationstation is up.

HerbFargus, 2016-03-05:
would a simple sleep perameter work as a workaround?

gizmo98, 2016-03-05:
Could be. But there are already many sleep statements in sixad. I had not enough time to test rpi3.

gizmo98, 2016-03-05:
I added a 5s sleep period between "bluetooth daemon stop" and bt_device_enable + sixad-bin start. My controller connects now after several cold boots.

joolswills, 2016-03-05:
I have a ps3 controller - Will test with jessie and see if I can come up with something.

joolswills, 2016-03-05:
sorry missed above post - if the sleep helps it might be enough.

bengus64, 2016-03-21:
BenWIson, by default it show now time out after ten minutes. I changed my to 5, you can do so by...

sudo nano /var/lib/sixad/profiles/default

Make sure these values are here...

enable_timeout 1
timeout_mins 5

save and exit.

Prindle19, 2016-04-11:
I actually needed to add:

added sixad --start & in rc.local so it will run sixad --start but will allow emulation station and everything else to boot up. 

But, i can now re-pair after a reboot!

pfaffenrodt, 2016-06-17:
Is pairing still an issue?  I have an rpi3 and an ps3 Controller. Controller works only with usb. Without usb connected LEDs not even blinking. Updated all today.

PeterPablo, 2016-06-20:
I also still have problems pairing with an up2date RetroPie setup, though did not yet go the way of starting from scratch.

pfaffenrodt, 2016-06-27:
the comment from @Prindle19 solved my problem

closed 2016-10-22 by joolswills